PlotJuggler/PlotJuggler #472: minimum step size? (milli -> micro)

dagar, 2021-06-29:
I'm somewhat new to PlotJuggler, so I apologize if I'm missing something, but is the minimum step (or resolution in general) milliseconds?

In PX4 I sometimes look at low level things (eg raw IMU data) where the time between samples is hundreds of microseconds, but in PlotJuggler I can't seem to step smaller than 0.001 seconds (1 millisecond). 

Possibly related I've also noticed zooming in on the y axis things break down below 0.001. Is this a bug or related to the same minimum size?

[image]

[image]

If this is fixable I'd be happy to take an a stab at it if you point me in the right direction.

dagar, 2021-06-30:
I believe it's the same with any ulg, but here's one I was looking at.

https://logs.px4.io/plot_app?log=b5c2fd73-b0d4-4fbf-9b6d-59cb6406caa2 (Download link upper right corner)

Is it possible to step through the timeline in smaller increments? For example `vehicle_imu` has samples every ~4000 us, but I can't set the step size below 0.01.

[image]

facontidavide, 2021-07-05:
ok, now I understand your problem.

Sorry, but I am not going to change this. Unfortunately there will always be a 1 % of user that will not be able to the default options and you are in that 1%

closed 2021-07-05 by facontidavide

dagar, 2021-07-05:
> Sorry, but I am not going to change this. Unfortunately there will always be a 1 % of user that will not be able to the default options and you are in that 1%

Fair enough, thanks for the response.