PlotJuggler/PlotJuggler #475: PlotJuggler for Foxy running on Raspberry Pi

thanhttran21, 2021-07-02:
Hello,

I am running Ubuntu 20.04 on a Raspberry Pi 4 with ROS2 Foxy installed on the Pi. I am looking to install PlotJuggler in order to do real-time plotting with my ROS topics. I have tried installing PlotJuggler in 3 ways:

1. Using `sudo apt install ros-${ROS_DISTRO}-plotjuggler` where I get the following error message:
`Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  binutils-dev comerr-dev krb5-multidev libgssrpc4 libkadm5clnt-mit11
  libkadm5srv-mit11 libkdb5-9 libkrb5-dev libnorm-dev libnorm1 libpgm-5.2-0
  libpgm-dev libsodium-dev libzmq3-dev libzmq5 ros-foxy-plotjuggler
  ros-foxy-plotjuggler-msgs
Suggested packages:
  doc-base krb5-doc krb5-user libnorm-doc
The following NEW packages will be installed:
  binutils-dev comerr-dev krb5-multidev libgssrpc4 libkadm5clnt-mit11
  libkadm5srv-mit11 libkdb5-9 libkrb5-dev libnorm-dev libnorm1 libpgm-5.2-0
  libpgm-dev libsodium-dev libzmq3-dev libzmq5 ros-foxy-plotjuggler
  ros-foxy-plotjuggler-msgs ros-foxy-plotjuggler-ros
0 upgraded, 18 newly installed, 0 to remove and 122 not upgraded.
Need to get 5579 kB/11.1 MB of archives.
After this operation, 48.9 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Err:1 http://packages.ros.org/ros2/ubuntu focal/main arm64 ros-foxy-plotjuggler arm64 3.1.0-1focal.20210423.013412
  404  Not Found [IP: 140.211.166.134 80]
Err:2 http://packages.ros.org/ros2/ubuntu focal/main arm64 ros-foxy-plotjuggler-msgs arm64 0.1.2-1focal.20210423.045523
  404  Not Found [IP: 140.211.166.134 80]
Err:3 http://packages.ros.org/ros2/ubuntu focal/main arm64 ros-foxy-plotjuggler-ros arm64 1.1.0-1focal.20210423.083848
  404  Not Found [IP: 140.211.166.134 80]
E: Failed to fetch http://packages.ros.org/ros2/ubuntu/pool/main/r/ros-foxy-plotjuggler/ros-foxy-plotjuggler_3.1.0-1focal.20210423.013412_arm64.deb  404  Not Found [IP: 140.211.166.134 80]
E: Failed to fetch http://packages.ros.org/ros2/ubuntu/pool/main/r/ros-foxy-plotjuggler-msgs/ros-foxy-plotjuggler-msgs_0.1.2-1focal.20210423.045523_arm64.deb  404  Not Found [IP: 140.211.166.134 80]
E: Failed to fetch http://packages.ros.org/ros2/ubuntu/pool/main/r/ros-foxy-plotjuggler-ros/ros-foxy-plotjuggler-ros_1.1.0-1focal.20210423.083848_arm64.deb  404  Not Found [IP: 140.211.166.134 80]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?`

2. I tried to install using the "How to build" section of the README.md file:
`git clone https://github.com/facontidavide/PlotJuggler.git
sudo apt -y install qtbase5-dev libqt5svg5-dev libqt5websockets5-dev libqt5opengl5-dev libqt5x11extras5-dev
mkdir build; cd build
 cmake ..
 make
 sudo make install`
I tried to enter `plotjuggler` in my terminal and I get `plotjuggler: command not found`

3. I tried to install using `sudo snap install plotjuggler` and I get this error: `error: snap "plotjuggler" is not available on stable for this architecture
       (arm64) but exists on other architectures (amd64).`

Could somebody please give me detailed steps on how to install PlotJuggler for my Raspberry Pi 4? (I'm a beginner :D)

Thank you so much for your time!

facontidavide, 2021-07-11:
I don't know about 1.

About error 2, look at where your binary is (should go to **/usr/local/bin**) 

Anyway, if you want to compile with ROS support, follow these instructions instead. 

https://github.com/PlotJuggler/plotjuggler-ros-plugins

facontidavide, 2021-07-11:
Take into account that you may want to disable OpenGL on your platform (see --help from command line)

closed 2021-07-11 by facontidavide