PlotJuggler/PlotJuggler #477: [ROS2/foxy]Segmentation Fault when trying to  topic Re-Publisher

galAcarteav, 2021-07-05:
Hey,
First of all, thank you for a great repo.
I build from source, everything looks fine except the ROS2 Topic Re-Publisher feature, when Im clicking on the check box the app is crash:
Stack trace (most recent call last):
#31   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f2e607a7564, in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)
#30   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7f2e5f6fc4a2, in g_main_context_iteration
#29   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7f2e5f6fc3ff, in 
#28   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7f2e5f6fc17c, in g_main_context_dispatch
#27   Object "/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5", at 0x7f2e5c0ee32d, in 
#26   Object "/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x7f2e60b1435a, in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>)
#25   Object "/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x7f2e60b3a10a, in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*)
#24   Object "/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x7f2e60b387d2, in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*)
#23   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f2e6074f939, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#22   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f2e6117a0ef, in QApplication::notify(QObject*, QEvent*)
#21   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f2e61170a65, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#20   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f2e611d21eb, in 
#19   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f2e611cf35c, in 
#18   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f2e61179456, in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool)
#17   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f2e6074f939, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#16   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f2e6117a342, in QApplication::notify(QObject*, QEvent*)
#15   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f2e61170a65, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#14   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f2e611b32b5, in QWidget::event(QEvent*)
#13   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f2e61267034, in QAbstractButton::mouseReleaseEvent(QMouseEvent*)
#12   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f2e61266e0f, in 
#11   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f2e61276914, in QCheckBox::nextCheckState()
#10   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f2e612672fc, in QAbstractButton::setChecked(bool)
#9    Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f2e61265ca5, in 
#8    Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f2e61265865, in QAbstractButton::toggled(bool)
#7    Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f2e6077b457, in QMetaObject::activate(QObject*, int, int, void**)
#6    Object "/home/gal/libs/ws_plotjuggler/install/plotjuggler_ros/lib/plotjuggler_ros/libTopicPublisherROS2.so", at 0x7f2e3e975640, in TopicPublisherROS2::setEnabled(bool)
#5    Object "/home/gal/libs/ws_plotjuggler/install/plotjuggler_ros/lib/plotjuggler_ros/libTopicPublisherROS2.so", at 0x7f2e3e974854, in TopicPublisherROS2::updatePublishers()
#4    Object "/opt/ros/foxy/lib/librclcpp.so", at 0x7f2e3f54970b, in rclcpp::PublisherBase::PublisherBase(rclcpp::node_interfaces::NodeBaseInterface*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rosidl_message_type_support_t const&, rcl_publisher_options_t const&)
#3    Object "/opt/ros/foxy/lib/librcl.so", at 0x7f2e3f742fff, in rcl_publisher_init
#2    Object "/opt/ros/foxy/lib/librmw_fastrtps_cpp.so", at 0x7f2e3f34cd71, in rmw_create_publisher
#1    Object "/opt/ros/foxy/lib/librmw_fastrtps_cpp.so", at 0x7f2e3f342869, in 
#0    Object "/opt/ros/foxy/lib/librosidl_runtime_c.so", at 0x7f2e3f3964ed, in get_message_typesupport_handle
Segmentation fault (Address not mapped to object [0x7f2e3e1a9220])

facontidavide, 2021-07-11:
Please check that both **plotjuggler** and **plotjuggler-ros** are up to date and confirm to be the respective versions.

I suggest doing **apt update / apt upgrade** first

TanatosSpirit, 2021-07-13:
@facontidavide @galAcarteav 

I got the same problem. 

My system: Ubuntu 20.04, ROS2 Foxy, 
Repos: PlotJuggler = 3.2.1, plotjuggler-ros-plugins = 1.5.0, plotjuggler_msgs = 0.1.2

I don't know why, but if in the file `/plotjuggler-ros-plugins/plugins/TopicPublisherROS2/generic_publisher.h` replace

```cpp
 48    auto type_support = rosbag2_cpp::get_typesupport_handle(topic_type, "rosidl_typesupport_cpp",
                             rosbag2_cpp::get_typesupport_library(topic_type, "rosidl_typesupport_cpp"));
```
by
```cpp
49     auto library = std::move(rosbag2_cpp::get_typesupport_library(topic_type, "rosidl_typesupport_cpp"));
50     auto type_support = rosbag2_cpp::get_typesupport_handle(topic_type, "rosidl_typesupport_cpp", library);
```
then it works.

I also noticed that the original version works with `/tf`, `/tf_static`, but does not work with `visualization_msgs/msg/MarkerArray`.

facontidavide, 2021-07-14:
Something that I find out recently is that republishing must be stopped if a new rosbag is loaded. If you keep it running, it will crash.

Is this maybe the bug you are observing?

TanatosSpirit, 2021-07-15:
@facontidavide @galAcarteav 
I'm observing the following bug:

https://user-images.githubusercontent.com/39188971/125736574-7f1569f1-b5a3-45f1-96b9-1f8ba1af7d96.mp4

facontidavide, 2021-07-15:
Ok. I am spending my vacations away from a computer. Once I am back, I will address this

autumn-eel, 2023-12-30:
I encountered a similar bug with ROS2 Iron. Have you fixed this problem?

facontidavide, 2026-09-20:
PlotJuggler 4 does not carry over the ROS2 topic re-publisher (StatePublisher) feature — it is an explicit non-goal of the rewrite — so this crash path no longer exists. Closing as won't-fix against the new architecture.

---
*PlotJuggler 4 is now the official release and the default branch of this repo; PlotJuggler 3 is end-of-life and no longer receives features or fixes. If this still affects you in PJ4, please reopen or file a fresh issue — it will be looked at.*

closed 2026-09-20 by facontidavide