PlotJuggler/PlotJuggler #478: [ROS] Bug: not consistent timestamp re-publish while playing a rosbag

tolomeis, 2021-07-05:
Hi everyone! Since one of the last updates, I found a little annoying bug involving the playback of rosbag files.
## Problem description
When I tried to run the bag with the ROS re-publisher and the "Keep original timestamp" option, I found that the timestamps published to the /clock topic are not consistent in time. 
To be sure I echoed the `/clock` topic, which you can find here, and you can see (from the third timestamp) that in some instants the time published by PlotJuggler goes "back in time", which raises the "ROS time moved backwards exception" whenever a node calls a timer or a sleep function. 

```---
clock: 
  secs: 1621527114
  nsecs: 729352933
---
clock: 
  secs: 1621527114
  nsecs: 729416943
---
clock: 
  secs: 1621527114
  nsecs: 720763291
---
clock: 
  secs: 1621527114
  nsecs: 739353045
```
I tried with a simpler rosbag and I couldn't reproduce the error, so I think it comes from having two or more different topics with stamped messages at different rates, but I'm not sure.
## Steps to reproduce
I'm running Ubuntu 20.04 with ROS Noetic and PlotJuggler 3.2.0. Here is a screenshot with the settings from the re-publisher. 
I attached the rosbag that I used to reproduce the bug, but as I said I found that the error comes with every rosbag with multiple stamped topics.
 [testbag_min.zip](https://github.com/facontidavide/PlotJuggler/files/6763432/testbag_min.zip)
[image]

I hope to be helpful, thank you!

facontidavide, 2021-07-11:
Hi,

I am not sure I see the problem here. Publishing uses the motion of the slider at the bottom or the timetracker (vertical red line) to publish.

If you move it from the right to the left, you move back in time!

What would be wrong is if the PLAY button trigger this error, but this doesn't happen in my tests, but If I use PLAY, this is my output:

```
---
clock: 
  secs: 1621527114
  nsecs: 719434867
---
clock: 
  secs: 1621527114
  nsecs: 720763291
---
clock: 
  secs: 1621527114
  nsecs: 729352933
---
clock: 
  secs: 1621527114
  nsecs: 729416943
---
clock: 
  secs: 1621527114
  nsecs: 731022875
---
clock: 
  secs: 1621527114
  nsecs: 739353045
---
clock: 
  secs: 1621527114
  nsecs: 739391614
```

closed 2021-07-11 by facontidavide

TonyWelte, 2021-07-23:
I too have encountered this issue. When loading a bag with multiple topics /clock occasionally goes back in time even when using only PLAY. This does not seem to occur when loading a single topic of the bag.
With the bag provided by @tolomeis I got the following result:

https://user-images.githubusercontent.com/12954392/126754653-840d6931-aa27-4eec-8ebe-44f4342de9ac.mp4

TonyWelte, 2021-07-23:
I think I identified the problem. The clock is published based on the message instance time rather than the bag time ([statepublisher_rostopic.cpp#L457](https://github.com/PlotJuggler/plotjuggler-ros-plugins/blob/1c8ed43ad486046286c5c1b9c5c6bbf10a5f8f30/plugins/TopicPublisherROS/statepublisher_rostopic.cpp#L457)). So even though the messages are correctly ordered in the bag, the clock can change based on the timestamps of the messages.
The clock time should be set with `consecutive_msg.at(index).x` instead of `msg_instance.getTime()`