PlotJuggler/PlotJuggler #479: Import of CSV with semicolon fails

maracuya-robotics, 2021-07-07:
# Problem description

After updating from 3.1.1 to 3.2.1 (ROS Noetic) CSV files with semicolon as delimiter cannot be loaded anymore. This is a regression from 3.1.1.

# Steps to reproduce (important)

- Download CSV file from #48
- Open in PlotJuggler. In the "Select the time axis" dialog only two items are available: `INDEX (auto-generated)` and the CSV header as a single item `;A;B;C;D`.

facontidavide, 2021-07-07:
yes, I am aware.

There are technical reasons to do that, in particular to avoid issues with the new plugin that export CSV files.

The solution is to make the delimiter configurable in the GUI.

facontidavide, 2021-09-05:
done!

[image]

closed 2021-09-05 by facontidavide