PlotJuggler/PlotJuggler #480: Plotting csv data series doesn't match what is expected.  Y-axis is off

jmailloux, 2021-07-07:
Thanks for contributing to PlotJuggler. You are great!

# Problem description
Plotting a data series from a csv file.  Something appears wrong with the y-axis.  Everything is shifted.

# Steps to reproduce (important)
- Plot "equivalent_ground_minus_wind_m_s" from attached csv file.
- Expected to get something like what was generated with Octave (attached)
- Got something that seems to be shifted on the y-axis.
- 
- Describe your platform / Operative System. - MacOS Big Sur.  11.4.
- Getting the same result using PlotJuggler -t and plotting _sin
[image]
[image]
[log_35_2021-6-30-17-00-56_airspeed_validated_0.csv](https://github.com/facontidavide/PlotJuggler/files/6779811/log_35_2021-6-30-17-00-56_airspeed_validated_0.csv)

facontidavide, 2021-07-11:
Unfortunately, it looks good to me

[image]

I do not support MacOS, sorry. Unless I get enough [Sponsors](https://github.com/sponsors/facontidavide) to buy a Mac, any issue related to MacOS will remain marked "wontfix".

Sorry :(

closed 2021-07-11 by facontidavide

jmailloux, 2021-07-19:
FYI, this seems to have magically fixed itself.  Maybe due to a reboot.