PlotJuggler/PlotJuggler #492: Remove Opening Memes

Adam-Antios, 2021-08-19:
PlotJuggler is a great project, but I have seen that when I suggest it to someone for professional use, the opening memes make them hesitant to consider it as a serious tool, but more of a gimmick. Beyond giving a negative first impression, memes don't serve a purpose, so would it be possible to be removed?

I use PlotJuggler and as the memes don't hinder my experience in any way, the only reason I mention this is because it is a shame for people to not use it after a bad first impression resulting from them.  Concluding, I want to thank everyone contributing to this project for their great work!

facontidavide, 2021-08-19:
Just use the option - n, when you start the application.

facontidavide, 2021-08-19:
To explain my point of view in more detail. It is possible to release a binary version with a "professional" splash screen.

Meme are something that made plotjuggler unique and memorable and for the time being, I am not planning to remove them. On the other hand, I might add the option to remove then in the settings menu.

Adam-Antios, 2021-08-19:
Thank you for your work and the quick response. As I mentioned I have no issue with the memes personally. Thy are fun! I just see them impede the adoption of an amazing software. The release of a more "professional" version would be amazing!

facontidavide, 2021-08-28:
I have been thinking about this.

You mention that memes can hurt the adoption in a **professional** context.
But if a company wants to use PlotJuggler, shouldn't they consider paying for it?

If there was a "professional" version of PJ. would YOUR company pay for it?

closed 2021-08-31 by facontidavide

ryanjAA, 2021-09-20:
Ya. Don’t change the memes. They’re funny and won’t hurt adoption at all. 

And if you really don’t want them then a paid version like @facontidavide suggests (if you don’t want to -n on start). I laugh every time. Especially when it catches me off guard.

saikishor, 2022-01-10:
Agreed! We love memes!