PlotJuggler/PlotJuggler #493: Can only load csv and ulg filers

natyshemer, 2021-08-24:
Hi, 

I have just switched from ubuntu 16 to ubuntu 18 and I can't get the plotjuggler to analyze my bag files. 
When i try to upload data i get only the *csv and *ulg files available 
[image]

I am using Ubuntu 18.04 and plotjuggler version 3.2.0
I did get the below  errors when running plotjuggler:

Thanks a lot and I absolutely love plotjuggler 

```
rosrun plotjuggler plotjuggler 
XmbTextListToTextProperty result code -2
XmbTextListToTextProperty result code -2
Loading compatible plugins from directory:  "/opt/ros/melodic/lib/plotjuggler"
"libDataLoadCSV.so" : is a DataLoader plugin
"libDataLoadULog.so" : is a DataLoader plugin
"libDataStreamSample.so" : is a DataStreamer plugin
"libDataStreamSample.so" ...but will be ignored unless the argument -t is used.
"libDataStreamUDP.so" : is a DataStreamer plugin
"libDataStreamWebSocket.so" : is a DataStreamer plugin
"libDataStreamZMQ.so" : is a DataStreamer plugin
"libPublisherCSV.so" : is a StatePublisher plugin
Number of plugins loaded:  7 

Loading compatible plugins from directory:  "/opt/ros/melodic/lib/plotjuggler_ros"
Number of plugins loaded:  0 

Loading compatible plugins from directory:  "/indoorRobotics/indoor_ws/devel/lib/plotjuggler_ros"
Number of plugins loaded:  0 

Loading compatible plugins from directory:  "/home/naty/.local/share/PlotJuggler"
Number of plugins loaded:  0 

XmbTextListToTextProperty result code -2
XmbTextListToTextProperty result code -2
XmbTextListToTextProperty result code -2
XmbTextListToTextProperty result code -2
XmbTextListToTextProperty result code -2
Gtk-Message: 13:41:30.955: GtkDialog mapped without a transient parent. This is discouraged.
XmbTextListToTextProperty result code -2
XmbTextListToTextProperty result code -2

```

facontidavide, 2021-08-24:
as described in the documentation, you must install:

          sudo apt install ros-melodic-plotjuggler-ros

If that doesn't work, let me know :)

closed 2021-08-24 by facontidavide

improve100, 2021-11-07:
sudo apt install ros-neotic-plotjuggler-ros  not found

aeudes, 2021-11-07:
ros-n**oe**tic-plotjuggler-ros should work better....