PlotJuggler/PlotJuggler #500: can't edit curves on xy plot mode

zhangshengping, 2021-08-30:
Thanks for contributing to PlotJuggler. You are great!

# Problem description

import data from csv file, and create a xy plot.
but i **can't edit curves**.
I build it from souce on main branch

# Steps to reproduce (important)

- import csv file
- create a xy plot 
- edit curves, but failed
[log440_akm_status_log_0.csv](https://github.com/facontidavide/PlotJuggler/files/7075958/log440_akm_status_log_0.csv)

facontidavide, 2021-08-30:
If I understand you correctly, you are right and that can not be done. XY plot are not new curves but a visual representation, therefore you can't use them as "input" for custom curves

closed 2021-08-30 by facontidavide

zhangshengping, 2021-08-31:
I'm sorry for my poor english. 
I want to display XY plot as dot and line , but I can't do this!
but it seem work on version 3.2.
[image]