PlotJuggler/PlotJuggler #501: Memory usage

Betschler, 2021-08-31:
# Problem description
When PlotJuggler is subscribing for to long, it runs out of memory. Resulting in a immediate crash of my Ubuntu system.
Behaviour is reproducible. Clearing all points or deleting all data does not have an impact on the memory usage.  Only restarting the application frees the memory.

# Steps to reproduce (important)
- I am using Ubuntu 20.04 with ROS Noetic.

facontidavide, 2021-08-31:
Can it be reproduce with Dummy Streamer? Are you using ROS streaming? ROS1 or ROS2?

Betschler, 2021-08-31:
ROS1 with "ROS Topic Subscriber"
As long as I subscribe the memory usage rises. If I pause or clean it it remains unchanged. If I let it consume all the memory my computer gets stuck and stops responding to anything. -> I have to force a reboot by the power button.

I am not sure: the Dummy Streamer is the" -t" Option? If so no. The problem occurs only during subscription to a ROS Topic

facontidavide, 2021-08-31:
Which version of PlotJuggler are you using? Does this affect master?

Betschler, 2021-08-31:
I am currently using 3.2.0. The ROS master? I do not think so, but I am not sure about that...

facontidavide, 2021-08-31:
I will investigate

Betschler, 2021-08-31:
Hi, somehow it works now. 
Haven't changed anything, but the memory usage isn't growing with the subscription anymore...
Strange... 
But for now, problem solved or at least not present anymore. :)

facontidavide, 2021-09-05:
I think I know what that is. Messages are stored in memory to allow re-publishing later.
This is dangerous if you subscribe to big messages such as images and point clouds. 

To play it safe, I decided to add a feature to the UI and set storing in Memory false by default.

[image]

facontidavide, 2021-09-05:
By the way, this change is NOT in this repository, but is part of https://github.com/PlotJuggler/plotjuggler-ros-plugins

closed 2021-09-05 by facontidavide