PlotJuggler/PlotJuggler #507: Can't add timeseries to plots in Windows (PlotJuggler 3.3.0)

Bartimaeus-, 2021-09-05:
Unable to add time series to plots in PlotJuggler 3.3.0 compiled for Windows:

https://user-images.githubusercontent.com/2954254/132111004-65d38844-bcef-4b8f-aaee-0632d49c1677.mp4

Same file works in Linux:

https://user-images.githubusercontent.com/2954254/132111010-e2fda361-b3b1-44d5-ad8d-410395704472.mp4

Test File attached:
[US06_drive_profile.csv](https://github.com/facontidavide/PlotJuggler/files/7110682/US06_drive_profile.csv)

facontidavide, 2021-09-05:
I am working to fix this. The entire 3.3.0 version seems to be broken on Windows

facontidavide, 2021-09-05:
fixed. thanks for reporting

[image]

closed 2021-09-05 by facontidavide