PlotJuggler/PlotJuggler #508: PlotJuggler 3.3.0 Crashes with CSV import using Rows as Time

Bartimaeus-, 2021-09-05:
Two pieces to this issue:
(PlotJuggler 3.3.0 build on Windows 10)

1. The "OK" button in the new dialog window for opening csv files is disabled when the "Use row number as X axis" option is selected unless a column entry is highlighted in the Column selection box on the left
2. After selecting a column entry and then the "Use row number as X axis" option and clicking the "OK" button PlotJuggler crashes when attempting to parse the file  

See the video below:

https://user-images.githubusercontent.com/2954254/132137284-ad8e9916-c6fa-4b12-add7-8a496284b3ad.mp4

Test csv file:
[US06_drive_profile.csv](https://github.com/facontidavide/PlotJuggler/files/7112132/US06_drive_profile.csv)

Bartimaeus-, 2021-09-05:
Also noting that the Cancel button is also disabled until a column is selected

facontidavide, 2021-09-05:
Because of the meme, you deserve a prompt solution to this issue

facontidavide, 2021-09-05:
I was able to reproduce the issue and solve it. It should be fixed now.

Thanks for reporting. I wrote the new CSV plugin in 1 day, and it is  probably a bit rough around the edges.

closed 2021-09-05 by facontidavide