PlotJuggler/PlotJuggler #509: Unable to Import CSV files with Strings Containing Commas

Bartimaeus-, 2021-09-05:
(PlotJuggler 3.3.0 on Windows)

PlotJuggler is unable to import CSV files that contain text strings with commas (text strings are surrounded by quotes)

The import fails differently depending on if the strings containing commas is in the header or the following rows.

**Header items that have comma-containing strings**
Quotes are ignored and additional columns are assumed:
[image]

This type of heater formatting is used by some data acquisition equipment such as the [Onset HOBO 4 Channel Thermocouple Logger](https://www.onsetcomp.com/products/data-loggers/ux120-014m/), which will include information like the channel number, units or settings in addition to the label.

**Data entries that  have comma-containing strings**
This will result in PlotJuggler giving an error that the number of entries in the data row is larger than expected from the header
[image]

One example of data that uses this formatting is from the online NOAA climate dataset, where the weather station name may contain a comma: https://www.ncdc.noaa.gov/cdo-web/search
Example data:
```
"STATION","NAME","LATITUDE","LONGITUDE","ELEVATION","DATE","TAVG","TMAX","TMIN","TOBS"
"USR0000CGOO","GOOSE VALLEY CALIFORNIA, CA US","33.0736","-116.8458","466.3","2021-01-01","13.2","18.3","10.0",
"USR0000CGOO","GOOSE VALLEY CALIFORNIA, CA US","33.0736","-116.8458","466.3","2021-01-02","11.7","17.2","8.3",
```
In this instance, the "NAME" data which contains a comma is not plottable and could be safely ignored. This does raise questions about support for string entries (if any is intended or would be supported) so that 1) string values could be viewed in the tree in the left hand column (would allow for viewing debug messages for example) and 2) string data could be preserved for csv export

Example file attached (note that some lines have blank entries for temperatures)
[2703382.csv](https://github.com/facontidavide/PlotJuggler/files/7112082/2703382.csv)

facontidavide, 2021-09-05:
Ok, I tried my best to address this special case with a custom SplitLine function.

I think it should work now.
For the records, data like the one you posted doesn't make much sense in PlotJuggler, since it seems to be particularly suited for SQL queries, rather than plotting.

About showing strings, this will be possible in the future.

facontidavide, 2021-09-05:
[image]

closed 2021-09-05 by facontidavide

facontidavide, 2021-09-05:
Done :) 

https://user-images.githubusercontent.com/2822888/132144147-3f3499a4-eb35-4d67-9168-0e4f346b56b4.mp4

Bartimaeus-, 2021-09-05:
Cheers mate
[image]

commit 665510a6 (2021-09-05): custom SplitLine function in CSV. Fix #509