PlotJuggler/PlotJuggler #510: [Feature Request] Moving RMS Filter

Bartimaeus-, 2021-09-06:
Request for a Moving RMS filter. 

Pretty much the same thing as the existing Moving Average filter but calculating root mean square instead:

[image]

This is useful for analyzing power and current, especially for data that changes sign like in motor drives with regen or AC power circuits

Bartimaeus-, 2021-09-11:
I've created an example csv dataset with Python that has a sine wave and applied moving RMS for window size of 10,25,100. I've also created a custom timeseries Lua script for moving RMS.

Screenshot below shows the original sine wave as well as the moving RMS from the Python script and the Lua script, with the difference in values for the two plotted in the lower chart, which looks small enough to be rounding error:
[image]

This should aid anyone who wants to do moving RMS or for testing implementation of the filter in PlotJuggler.

@facontidavide feel free to bundle that Lua function with PlotJuggler if it's up to snuff and desired

Files are attached:
[PlotJuggler_movingRMS_Test.zip](https://github.com/facontidavide/PlotJuggler/files/7148353/PlotJuggler_movingRMS_Test.zip)

facontidavide, 2021-09-20:
Done :) 

[image]

[image]

closed 2021-09-20 by facontidavide