PlotJuggler/PlotJuggler #511: [Feature Request] Use Filtered Data as a new Timeseries

Bartimaeus-, 2021-09-06:
It would be nice to be able to treat filtered data as a new timeseries that could be used for the following:
-Input to custom time series (apply a filter and then run it through a custom function). You could also do [custom series->apply filter->use as input to another custom series]
-Apply multiple filters sequentially to data (ex: apply a scale and then a moving average)

The main things that I am running into right now is the need to scale data to match units. I have custom functions that I want to apply to different datasets and sometimes data is given as milliVolts or Volts, milliAmps or Amps, etc. and I'd like to calculate power as well as watthours, and scaling the data as an input so the units all match simplifies the reuse of functions

The other main use case I have for this is applying a moving RMS filter to the data and then running it through a custom function so you can calculate the max RMS power for a given duration in a dataset. (apply moving RMS filter, then put it through a custom function to get the max value)

Treating them as a new time series would also have the benefit of being able to view the filtered values in the left hand column, but that seems a much lower priority
[image]

facontidavide, 2021-09-19:
I think it is a good idea, but it will take time before I implemnt this

Bartimaeus-, 2021-10-09:
I've implemented basic functionality for the feature:

https://user-images.githubusercontent.com/2954254/136670890-59b3b389-97ca-4c83-990b-f16c79124c7e.mp4

Some notes on the feature:
-Curves are only copied if a transform has been applied
-Curves that are hidden on the plot are not copied
-There should be a better way to modify the TransformedTimeseries class 
-Names of the copies are the original series name with "_copy" added. Future change could be made to allow user customization of the name of the new series

The code can be found here, I can make a pull request if it looks good
https://github.com/Bartimaeus-/PlotJuggler

facontidavide, 2022-05-14:
The problem I see with your implementation is that the new curve is not saved in the Layout files.

To do it "right", I need a large refactoring of some internal class hierarchies.

Stay tuned.

Bartimaeus-, 2022-05-16:
A limitation to the way I did it is that it only copies the contents from the plot into the new series and there is no dynamic link from the new series to the old one. Giving access to the built-in transformations within the custom series Lua scripts might be simpler and more powerful than simply copying the data from the plot into a new series. This essentially is what Excel does with the functions that are callable in custom formulas

Example workflow:
[image]

Apply transformations:
[image]

Output is a custom series with a call to the correct lua function (user could also use these functions however they want in a custom Lua script)
[image]

facontidavide, 2022-05-16:
The problem here (maybe a bad decision) is that those transforms are treated differently than the one generated with Lua.

The solution you propose is valid, but I think I can do better.

I want to do this change, but to make it "right" I need to think carefully :smile:

ebnerluca, 2025-01-09:
Any updates on this?

I would love to see such a feature which allows a workflow to apply multiple transforms sequentially (e.g. moving average + derivative).
This could be done by either:
- applying a single transformation, saving the new timeseries, dragging it into the plot and applying the next transform (repeat this apply-save-apply workflow if you want more than two transforms).
- or directly allowing selecting multiple sequential transforms in the transform editor with stating their order

facontidavide, 2026-09-20:
PlotJuggler 4's Data Processors turn each filter/transform into a real, chainable timeseries in the datastore (visible in the curve list, saved in layouts, and usable as input to further transforms or scripts) — exactly the sequential/reusable workflow requested. Closing as delivered in PJ4.

---
*PlotJuggler 4 is now the official release and the default branch of this repo; PlotJuggler 3 is end-of-life and no longer receives features or fixes. If this still affects you in PJ4, please reopen or file a fresh issue — it will be looked at.*

closed 2026-09-20 by facontidavide