PlotJuggler/PlotJuggler #513: How to filter time series ros topic elements

ArduinoHocam, 2021-09-10:
Hi, 
I think this will be mostly feature request. However, I'm sure that many people need this feature just like me.

In the plotjuggler, Under the Time Series List,  some topics may have many sub elements. Each element has same attribute like id, timestamp etc. We can type and search the corresponding topic or element attribute like : "my_topic target id". It is hard to search by hand and find the relevant id by clicking drop and down button. 
My question is : Is there a way to filter and search the time list as follows:
"my_topic.target.id==10" ? or maybe some regEx operations? 
Thanks.

facontidavide, 2021-09-10:
With the current implementation, this feature would be quite hard to achieve.

I don't see this feature being implemented any time soon.

If a company want to speed up or prioritize the implementation of this feature, consider sponsoring its development

facontidavide, 2021-09-19:
More details here: https://github.com/facontidavide/PlotJuggler/discussions/520 :wink:

closed 2021-09-19 by facontidavide