PlotJuggler/PlotJuggler #517: Qt5 directory on macOS Big Sur (M1)

erickisos, 2021-09-19:
Hi!

First of all, this is not a problem with PlotJuggler.

# Problem description

I recently cloned the repo and when I was trying to compile it, I get an error message saying that Cmake can't find Qt5.
I solved this using Qt5_DIR flag pointing to homebrew dir (`/opt/homebrew/qt@5/`), anyway, I read in the `CMakeLists.txt`  that in fact, exist a flow control for APPLE which check for `/usr/local/opt/qt5` so I can write the change to make it work by default on my M1, but I need to confirm, does anyone else with macOS confirm that the default installation on homebrew is that dir? I need some confirmation about that before making any changes to the CMake file.

# Steps to reproduce (important)

- macOS Big Sur (Apple M1)
- You can reproduce this using the default `cmake` on the same platform.

ryanjAA, 2021-09-20:
I think this would be environment specific since I can build on bigsur but was getting zmq.h errors until I simply copied it manually and rebuilt. I installed at via the qt installer guy everything else via brew. It crashes a lot though on Mac. Tried main (master), 3.2 release, etc so it’s likely just that this is lest tested in mac. 

As for your specific question, I added the brew directories to my path (via the zshrc) to get it to compile but that’s needed for most thing brew repeated.

PymZoR, 2021-10-30:
> I think this would be environment specific since I can build on bigsur but was getting zmq.h errors until I simply copied it manually and rebuilt. I installed qt via the qt installer gui and everything else via brew. It crashes a lot though on Mac. Tried main (master), 3.2 release, etc so it’s likely just that this is less tested on mac.
> 
> As for your specific question, I added the brew directories to my path (via the zshrc) to get it to compile but that’s needed for most thing brew related.

For the zmq.h error, I did the following to make it work : 
`cmake .. -DCMAKE_CXX_FLAGS="-I$(brew --prefix zmq)/include"`

On macOS Monterey the following path is used `/opt/homebrew/opt/zeromq`

closed 2021-11-08 by erickisos

mengchaoheng, 2023-03-07:
@PymZoR @erickisos can you help me? about this [issue](https://github.com/facontidavide/PlotJuggler/pull/725#issuecomment-1438769494)