PlotJuggler/PlotJuggler #521: Building on Windows

Georacer, 2021-09-19:
# Problem description
Cannot manually compile for Windows.

I want to create the [Ardupilot plugin for windows](https://github.com/khancyr/plotjuggler-apbin-plugins). This means that I have to build it from source on Windows. I am using Cygwin and its Cygwin Prompt to get the compiler, required packages and call cmake.
The plugin is a cmake project and it needs access to Plotjuggler's libraries, as per [this line](https://github.com/khancyr/plotjuggler-apbin-plugins/blob/main/CMakeLists.txt#L71) in the CMakeLists.txt.

However, this assumes that Plotjuggler's headers and libraries have been installed locally (i.e. Plotjuggler has been built and installed, not downloaded as an executable).

So, my question is, how can I build PlotJuggler from source so that I can point the plugin to the generated header files and the libraries? What are the prerequisites for building? I couldn't find any build instructions around.
I saw the `wind32build.bat` file but it doesn't work for me out of the bat, it gets stuck on number selection.

Thanks in advance for your time!

facontidavide, 2021-09-20:
Hi,

I do not maintain **wind32build.bat**, since that was contributed by a user with a PR.
Honestly, it doesn't work for me either, unless I manually modify it.

I created a page that can give you some hints about how to compile in Windows: https://github.com/facontidavide/PlotJuggler/discussions/518

>So, my question is, how can I build PlotJuggler from source so that I can point the plugin to the generated header files and the libraries?

Since you are using Cygwin, nothing prevents you from following the usual cmake/make/make install procedure, I think.

Your readme itself provides those instructions.

> What are the prerequisites for building?

The prerequisite for building are mostly related to Qt, this is the reason that using the QtCreator + Mingw will be more straightforward, IHMO.

 > I couldn't find any build instructions around.

Arguably, PlotJuggler is compiled as ANY generic **Qt + cmake** app. I don't believe that the purpose of the documentation is to teach this pre-required knowledge.

I think it might be valuable, in the future to consider using [Conan](https://conan.io/), but that is something I have no time to address right now..

Cheers

facontidavide, 2021-09-20:
Side reminder that, if the community of PlotJuggler users need better support, tutorials and faster bug fixing, they should consider [sponsorship](https://github.com/sponsors/facontidavide)

closed 2021-09-20 by facontidavide