PlotJuggler/PlotJuggler #522: Quits on Layout Load

ryanjAA, 2021-09-20:
Every time I load a layout PJ quits (and for good measure I made a layout based on building the Main branch [as of today[).

If I open PJ without loading a layout, it works for awhile and then on doing some specific transformations causes it to crash (non-specific, differs each time). 

Heres the kicker - this is on a Mac but im seeing similar (not exact) issues on Ubuntu. The similar being that specific transformations cause it to crash but not so much the quit on layout load (I know... two specific issues).

I gather this will be difficult to diagnose (given not officially supported on Mac) but willing to help however needed - something in the libraries when on a Mac.

Hands down the best data manipulation software available. :)

Here's the Mac output:

`Stack trace (most recent call last):
#31   Object "CoreFoundation", at 0x7fff204e9634, in __CFRunLoopDoSources0 + 242
#30   Object "CoreFoundation", at 0x7fff204e98b4, in __CFRunLoopDoSource0 + 180
#29   Object "CoreFoundation", at 0x7fff204e994c, in __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
#28   Object "libqcocoa.dylib", at 0x1099e0f98, in qt_plugin_instance + 199080
#27   Object "QtGui", at 0x108cc213b, in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 203
#26   Object "QtGui", at 0x108cdc329, in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) + 1385
#25   Object "QtGui", at 0x108cdcbdf, in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) + 3615
#24   Object "QtCore", at 0x1094912b6, in QCoreApplication::notifyInternal2(QObject*, QEvent*) + 166
#23   Object "QtWidgets", at 0x108700e02, in QApplication::notify(QObject*, QEvent*) + 466
#22   Object "QtWidgets", at 0x1086ffa56, in QApplicationPrivate::notify_helper(QObject*, QEvent*) + 262
#21   Object "QtWidgets", at 0x1087563de, in QDesktopWidget::qt_metacall(QMetaObject::Call, int, void**) + 4622
#20   Object "QtWidgets", at 0x108757ca5, in QDesktopWidget::qt_metacall(QMetaObject::Call, int, void**) + 10965
#19   Object "QtWidgets", at 0x108700362, in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) + 898
#18   Object "QtCore", at 0x1094912b6, in QCoreApplication::notifyInternal2(QObject*, QEvent*) + 166
#17   Object "QtWidgets", at 0x108702944, in QApplication::notify(QObject*, QEvent*) + 7444
#16   Object "QtWidgets", at 0x1086ffa56, in QApplicationPrivate::notify_helper(QObject*, QEvent*) + 262
#15   Object "QtWidgets", at 0x108738d0a, in QWidget::event(QEvent*) + 442
#14   Object "QtWidgets", at 0x1087e7e6c, in QAbstractButton::mouseReleaseEvent(QMouseEvent*) + 268
#13   Object "QtWidgets", at 0x1087e6ca9, in QAbstractButton::isCheckable() const + 457
#12   Object "QtWidgets", at 0x1087e6e2f, in QAbstractButton::isCheckable() const + 847
#11   Object "QtCore", at 0x1094c201d, in QMetaObject::activate(QObject*, QMetaObject const*, int, void**) + 3373
#10   Object "plotjuggler", at 0x107b5369e, in MainWindow::qt_metacall(QMetaObject::Call, int, void**) + 78
#9    Object "plotjuggler", at 0x107b990b8, in MainWindow::on_pushButtonLoadLayout_clicked() + 408
#8    Object "plotjuggler", at 0x107b82430, in MainWindow::loadLayoutFromFile(QString) + 6064
#7    Object "plotjuggler", at 0x107be134e, in LuaCustomFunction::xmlLoadState(QDomElement const&) + 30
#6    Object "plotjuggler", at 0x107be02d2, in LuaCustomFunction::initEngine() + 834
#5    Object "plotjuggler", at 0x107be3026, in void sol::basic_reference<false>::move_assign<false>(sol::basic_reference<false>&&) + 38
#4    Object "plotjuggler", at 0x107c52aa8, in luaL_unref + 40
#3    Object "plotjuggler", at 0x107c2ee90, in lua_rawgeti + 80
#2    Object "libsystem_platform.dylib", at 0x7fff20435d7c, in _sigtramp + 28
#1    Object "plotjuggler", at 0x107c1b1dd, in backward::SignalHandling::sig_handler(int, __siginfo*, void*) + 13
#0    Object "plotjuggler", at 0x107c1b242, in backward::SignalHandling::handleSignal(int, __siginfo*, void*) + 66
zsh: segmentation fault  plotjuggler`

tomcat327, 2021-10-21:
I'm seeing similar issues on Ubuntu with `ros2 run plotjuggler plotjuggler -d <bag_file> -l <layout_file>`

facontidavide, 2021-10-29:
@tomcat327 and @ryanjAA  please share the bug file and layout file with me

facontidavide, 2021-10-30:
Latest commit on **main** branch ( 2ca9a91bd7788a5f7a8528826ef7d1c9fc39c349 ) should fix this. Can you please confirm?

vj2181, 2021-12-01:
I have PJ 3.3.2 and I am primarily using it for ROS2 bags. There seems to be an issue loading layouts when custom series are involved. As an example I have attached a small ROS2 bag file with random data and explained the steps I took below. 
[Zip file with ROS2 bag and PJ layout files](https://github.com/facontidavide/PlotJuggler/files/7637337/test_ros2bag_and_layout_files.zip)

I loaded the ROS2 bag data and plotted the series and saved the layout (layout_without_custom_series.xml). After closing plotjuggler, we can reopen the ROS2 data bag and recreate the plot by loading the layout_without_custom_series.xml file. The resulting screenshot is below. Everything seems to work fine.
[image]

Then I restarted PJ, reopen the ROS2 bag plot the regular series, and also create a custom series called "half_Series" and created the plots shown below. This layout was saved as layout_with_custom_series.xml.
[image]

Then I restarted PJ, reopened the ROS2 bag. Then trying to load the layout results in PJ hanging with the below error shown on the screen.
[image]

At times an error like the below also shows up when loading layouts with custom series
[image]

zm2022, 2021-12-02:
Thank you for developing and sharing plotjuggler - as others have said, it has significantly sped up my rosbag data analysis.

I'm using it to work with ROS2 bags on Ubuntu 18.04.

I'm working with Plotjuggler version 3.3.3, and am also experiencing this issue - when I open a layout that has no custom series saved, the data and plots display correctly. However, when I try opening a layout that has a custom series saved in it, I run into issues - if I am trying to open a custom series layout that was created with the "save data source" box checked, plotjuggler immediately closes, and I get this error message:
[image]

If I try opening a custom series layout that didn't have the "save data source" box checked, I run into similar issues as posted by user vj2181.

facontidavide, 2021-12-02:
@vj2181 thanks for taking the time to describe the process.
@zm2022 are you running on Mac too?

facontidavide, 2021-12-02:
I reported this to the creator of Sol3, since the problem seems to be there.

@vj2181 
@zm2022 
@ryanjAA 
@tomcat327 
@jmailloux

Can you please try branch **lua_fix** and tell me if that solve the issue?

zm2022, 2021-12-02:
@facontidavide , I am working on Ubuntu 18.04, not mac. I'll try the lua_fix branch and let you know. Thanks!

zm2022, 2021-12-03:
@facontidavide , I installed the lua_fix branch, and so far that has solved the issue. On Ubuntu 18.04, I am able to export and import layouts with custom series, with no crashing or error messages anymore. Thank you so much!!

jmailloux, 2021-12-05:
@facontidavide, this solves the issue on MacOS.  Thanks!

ryanjAA, 2021-12-05:
Will give the branch a try.

vj2181, 2021-12-06:
This fix also solved my layout load issue when using custom series under ROS2 for Ubuntu 20.04. Thank you.

maracuya-robotics, 2021-12-07:
Had also crashes with loading custom timeseries from a layout file and the `lua_fix` branch has fixed it. Thank you!

ryanjAA, 2021-12-14:
lua_fix branched fixed it for me!

Thanks!

Side note, @facontidavide let's get us users together to get you a Mac (iirc you said in another post you didn't have one and if we all wanted to fund one you would actively support it). Can you start a funding page (if you're still interested in doing that)?

JohannesPankert, 2021-12-23:
Has the 'lua_fix' branch been merged to main? I have experienced the same bug with the compiled version from the `noetic` PPA but not with the current version on `main`. Are you planning to create a new release soon?

closed 2022-01-29 by facontidavide