PlotJuggler/PlotJuggler #524: Opening a CSV File With Duplicate Header Entries Makes PlotJuggler Crash

Bartimaeus-, 2021-09-25:
Attempting to open a csv file that contains two or more header entries with the same contents causes PlotJuggler to crash without warning

Example header:
`timestamp,A,A,C,D`

Attached is the command line output when PlotJuggler crashes:

[PlotJuggler_CSV_MultipleHeader_Crash_Print_Outputs.txt](https://github.com/facontidavide/PlotJuggler/files/7230341/PlotJuggler_CSV_MultipleHeader_Crash_Print_Outputs.txt)

Having PlotJuggler check the header for duplicate entries and either aborting the file load with an informational message or appending a sequential number to the end of the duplicate entries to make them unique would be great.

Having non-unique header entries in a csv is definitely not good practice, in this case the csv file that caused this issue was generated from a data analysis software that was exporting separate time columns for each data series:

`Time,Data1,Time,Data2,Time,Data3`

Multiple time axis from the same csv is definitely not a format that PlotJuggler currently supports, the intent was to ignore the time columns and use the row number as the time axis.

facontidavide, 2021-09-25:
Nice catch. I will fix it soon.

The correct behavior is, in my opinion, to simply forbid column with identical names (and stop loading such files).

In the  example you provided` Time,Data1,Time,Data2,Time,Data3`, an option could be to append some automatic suffix to identical columns, but i don't think plotjuggler should handle this.

facontidavide, 2021-10-01:
[image]

commit 93121f7a (2021-10-01): Fix #524 and #529
closed 2021-10-01 by facontidavide