PlotJuggler/PlotJuggler #525: Slashes in CSV Header Entries Cause Nested Tree Entries in Timeseries List

Bartimaeus-, 2021-09-25:
Slashes in csv file headers cause PlotJuggler to create nested tree entries in the Timeseries List in the left panel

Example header:
`timestamp,A,B,KM/H,D`

[image]

Funny enough this doesn't have a visible impact on the display of the name in the plot legend because all of the slashes are already displayed:
[image]

Bartimaeus-, 2021-09-25:
I just discovered that if you use the csv exporter the header will have the following format:

`__time,simple_parse_timestamp/A,simple_parse_timestamp/B,simple_parse_timestamp/D,simple_parse_timestamp/KM/H`

which is displayed in the tree as:
[image]

Which seems like an interesting way for maintaining the original tree format when opening a csv file that was created from PlotJuggler-- is this an intended feature that I've mistaken for a bug? Perhaps PlotJuggler could use the backslash ( \ ) or pipe ( | ) for this purpose instead of the forward slash ( / ) which is more likely to be used to indicate the unit of data in a header context?

facontidavide, 2021-09-25:
slashes are by definition THE way tree view is defined.

I understand that in there could be corener cases like this one, but all the existing plugins (including third party) use this.

You can use this potion, to avoid the tree view: 

[image]

closed 2021-09-25 by facontidavide