PlotJuggler/PlotJuggler #530: Plots get blurred

kminoda, 2021-09-27:
The plots on plotjuggler always get blurred on my PC. Please look at an uploaded video as it's difficult to explain in words.
I am using ROS2 Galactic on Ubuntu 20.04 LTS, running the app using `ros2 run plotjuggler plotjuggler`.

I am sorry that I cannot reproduce this problem on the other PCs... nobody around me (using plotjuggler on ROS/ROS2) have ever encountered this issue either.

Thanks in advance!

https://user-images.githubusercontent.com/44218668/134861902-0ac9d199-20b9-4630-a66a-601924bda6ed.mp4

--
Here are some screenshots for a reference. The first figure shows the blurred plot. For comparison, I also uploaded a plot with the less blur-ish effect.
[image]
[image]

facontidavide, 2021-09-27:
disable OpenGl in Settings

closed 2021-09-27 by facontidavide

kminoda, 2021-09-27:
It worked! Thanks for the quick response.