PlotJuggler/PlotJuggler #532: Outlier Removal Transform Does Not Respond to User Input

Bartimaeus-, 2021-09-28:
It appears that the outlier removal filter does not make use of the Max Jump parameter. 

See the video clip below:

https://user-images.githubusercontent.com/2954254/135173156-50ac3fca-ff8e-4687-9915-c3e64fd34334.mp4

To make a test file that can be used to replicate the issue I scattered some outlier points of value -200, -50, and 50 to a sine wave with amplitude of 1. All of the points were filtered regardless of the setting of Max Jump.

Additionally, no matter what input you set, the preview of the trace doesn't change and when you re-open the transform editor window the value is always 100.0

Test File
[generated_sine_rms_data_with_outliers.csv](https://github.com/facontidavide/PlotJuggler/files/7247548/generated_sine_rms_data_with_outliers.csv)

facontidavide, 2021-09-29:
The filter works just fine, simply, to make it "fail" you need to use a very big number such as 10.000 !

I will add some comments about the algorithm used

closed 2021-09-29 by facontidavide

facontidavide, 2021-09-29:
Fixed! Thanks for reporting