PlotJuggler/PlotJuggler #536: [ROS/noetic Bug] Lag on plot

Julien-Livet, 2021-10-06:
I discovered and used this amazing tool on recent robotic projects with ROS/melodic (on VM) !

# Problem description

I try it on ROS/noetic after ROS/melodic and I observe a "lag" on plots with ROS topic (I reproduce the problem with PJ default data).

[image]

# Steps to reproduce (important)

- Platform / Operative System: Xubuntu 20.04 / ROS noetic (on VM with VirtualBox)
- rosrun plotjuggler plotjuggler -t
- Select "Dummy Streamer" and plot "data_vect/0"
- Play (and pause) to observe lag

facontidavide, 2021-10-06:
Duplicate of #530 

[image]

Disable OpenGL as described here: https://github.com/facontidavide/PlotJuggler/issues/530

closed 2021-10-06 by facontidavide

Julien-Livet, 2021-10-06:
Thanks!
Sorry for duplication, I only saw open issues --'...