PlotJuggler/PlotJuggler #553: [ROS-noetic] I can't find the format for rosbag to import data!

Lenny-Lee-ustb, 2021-10-30:
# Problem description

I can't find the format for rosbag to import data

# Steps to reproduce (important)

ubuntu20.04&noetic
I can only find *.csv and *.ulg format in Files->Data->import.

facontidavide, 2021-10-30:
Dear User,

you probably:
- ignored the Message "Missing package [plotjuggler-ros]" in the popup at the beginning,
-  you haven't read the readme and 
- you haven't looked at previous solved issues.

Please do: 

`sudo apt install ros-noetic-plotjuggler-ros`

closed 2021-10-30 by facontidavide

Lenny-Lee-ustb, 2021-11-22:
> Dear User,
> 
> you probably:
> 
> * ignored the Message "Missing package [plotjuggler-ros]" in the popup at the beginning,
> * you haven't read the readme and
> * you haven't looked at previous solved issues.
> 
> Please do:
> 
> `sudo apt install ros-noetic-plotjuggler-ros`

I solved this after your guidence. 
The software is pretty good and efficient. 
Thanks a lot!!