PlotJuggler/PlotJuggler #557: Can't load a layout with a CSV file with semicolon as delimiter

VictorCampillo, 2021-10-31:
If I try to load a layout with a CSV file in which the delimiters are semicolons, the load fail with the next error

[image]

My OS is Xubuntu 20.04 64 bits, and I compile Plotjuggler from sources, the test was done with the latest revision 3.3.3 

To reproduce the test, just edit the file "simple_parse_timestamp.csv" and change the delimiters to semicolons.

I don't know if it could be related to the issue, but in the layout file I don't see any field that indicate the CSV delimiter, maybe it always guess that the delimiter is a comma.

[datasamples.zip](https://github.com/facontidavide/PlotJuggler/files/7448636/datasamples.zip)

facontidavide, 2021-10-31:
[image]

Well, it works just fine on my side

[image]

VictorCampillo, 2021-10-31:
On my side the dialog "CSV Loader" only shows if I specific the data file with -d. If I open Plotjuggler with the parameters -d and -l, then it works.

But if I open Plotjuggler only with the layout option, and the data file is set in the option "previouslyLoaded_Datafiles", then I get the error "Can't find one or more curves"

facontidavide, 2021-10-31:
Good news, problem fixed. Thanks for reporting!

But the next time, in steps to reproduce, don't say:

> To reproduce the test, just edit the file "simple_parse_timestamp.csv" and change the delimiters to semicolons.

But:

> Modify the file location of "simple_parse_timestamp.xml" and launch plotjuggler wil the option 
> `plotjuggler -l simple_parse_timestamp.xml`.

This would have saved me some time

facontidavide, 2021-10-31:
To test it, use this file, because the delimiter is now saved in the XML.

[simple_parse_timestamp_semicolon.zip](https://github.com/facontidavide/PlotJuggler/files/7449323/simple_parse_timestamp_semicolon.zip)

closed 2021-10-31 by facontidavide

VictorCampillo, 2021-10-31:
Thank you very much for the fix, now it's working.

Next time I will be more careful in the instructions to reproduce the issue.

Congratulations for your program and keep up the good work.