PlotJuggler/PlotJuggler #560: Segmentation fault on exit (M1 chipset)

erickisos, 2021-11-08:
# Problem description

I built PlotJuggler today for first time, and I discovered that every time I close the window (through close button) it always launch a `segmentation fault` like in the screenshot.

[image]

# Steps to reproduce (important)

It's important to consider a few things because it could be an error of the platform and not necessarily the application.

- macOS Big Sur 11.6 on Apple M1
- I don't have any ros instance installed (mainly because the lack of support for the platform(?))

I can share the full output if needed, I just want to be sure if this is a problem for the platform or if it's happening with the new version of PJ.

facontidavide, 2022-01-29:
This is fixed in Protobuf, but VideoViewer is still crashing :(

facontidavide, 2022-01-29:
fixed

commit 1465a430 (2022-01-29): fix #560
closed 2022-01-29 by facontidavide