PlotJuggler/PlotJuggler #561: PlotJuggler in ROS2 Launch file

gurbain, 2021-11-08:
# Problem description

I am struggling to start plotjuggler in a ROS2 launch file. It seems that starting a Node in a launch file automatically passes the *--ros-args* argument and plotjuggler does not handle it. Is there a way to circumvent this, while using the Node class of ROS2 launch actions (and not ExecuteProcess)? Thank you!

# Steps to reproduce 

- Platform: ROS2 Foxy on Ubuntu 20.04
- Sample launch file:
```python
from launch import LaunchDescription
from launch_ros.actions import Node

def generate_launch_description():
    node_plot = Node(
        package='plotjuggler',
        executable='plotjuggler')
    return LaunchDescription([node_plot])
```
- Behavior:
```sh
$ ros2 launch plot plot.launch.py
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [plotjuggler-1]: process started with pid [21577]
[plotjuggler-1] QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
[plotjuggler-1] Unknown option 'ros-args'.
[ERROR] [plotjuggler-1]: process has died [pid 21577, exit code 1, cmd '/opt/ros/foxy/lib/plotjuggler/plotjuggler --ros-args'].
```

facontidavide, 2021-11-08:
no, there isn't

GuihVicentini, 2022-04-22:
Hi @facontidavide I know this has been closed, but I am facing the same issue, if there is no solution, what would you advise to do if I would like to launch plotjuggler with a launch file in ROS 2?

frankkusters, 2022-05-05:
Might the [removeROSArgs() (ROS1)](http://docs.ros.org/en/noetic/api/roscpp/html/namespaceros.html#aa044af6270e37edd78cf05f5fc7ee148) or [remove_ros_arguments() (ROS2)](https://docs.ros2.org/galactic/api/rclcpp/namespacerclcpp.html#ad4968a767791995c011c57be22cd40c8) functions be of help? As the name says, they remove ROS arguments, allowing the code after it to work without the ROS arguments being in the way.

facontidavide, 2022-05-05:
I did not know about that. Sound as the right solution

facontidavide, 2022-05-14:
solution pushed

closed 2022-05-14 by facontidavide