PlotJuggler/PlotJuggler #562: Multi Y axis

hogAlert, 2021-11-08:
Thanks for writing and open-sourcing to PlotJuggler. You are great!

I don't know how many people need this, but one thing I really value (and have to go through many hoops to try and resolve) is being able to visualise multiple series each on their own Y axis (due to widely varying scales e.g engine speed 0-5000rpm, engine coolant temperature -40 to 120C, exhaust temperature 0 to 1000C, correction factor 0 to 1 etc.).
At present this is only possible either by factoring each one somehow (then remembering that the displayed value is not 'real') or making multiple plots (which is OK sometimes but not others).

I realise that this may be a big ask, put it on the bottom of your ideas pile where it will never be reached perhaps ;)

facontidavide, 2021-11-15:
not going to happen... ever.

Sorry :(

closed 2021-11-15 by facontidavide

edobezabb, 2023-11-24:
> not going to happen... ever.

Is this still the current position?
Is there any technical motivation for that or is it just not on the roadmap?

In our team we are evaluating a software for visualizing plot and this would be a needed feature. We are also open to contribute and make this possible, unless it's really very demanding feature to develop.