PlotJuggler/PlotJuggler #567: In newest version, layout clears itself everytime we open a rosbag

roangel, 2021-11-17:
So, I am using plotjuggler for around a year and it has been awesome, it makes us save TONS of time when trying to debug a problem. Thanks a lot for this tool.

So, regarding my problem, usually I have different rosbag files that contain the exact same topic names, but different data inside. Before, in version 3.2.0, when I had a layout with some data plotted, then switched to a new rosbag file with same topic names but different data, it just smoothly kept the layout and just changed the data. This was extremely convenient for debugging. However, in the latest version, when I change the file, the layout completely goes blank and then I have to, every single time, bring up the signals I wanted to analyze.

It would be really nice if this could go back to how it was before.

Thanks a lot!

roangel, 2021-11-18:
Yes, it's 3.3.3. Just checked and it definitely happens.

facontidavide, 2021-11-19:
Can you please send me a sample data/layout. It will make my life much easier :)

roangel, 2021-11-19:
Here is a sample bag that you can test with:
[test_plotjuggler.zip](https://github.com/facontidavide/PlotJuggler/files/7569709/test_plotjuggler.zip)

roangel, 2021-11-19:
Sample layout:
[test_plotjuggler_layout.zip](https://github.com/facontidavide/PlotJuggler/files/7570124/test_plotjuggler_layout.zip)

facontidavide, 2022-01-29:
I think it is fixed on master. Can you check?

closed 2022-01-29 by facontidavide