PlotJuggler/PlotJuggler #575: PJ build does not contain built-in plugins and crashes when clicking onto "plugin specific options"

patzf, 2021-12-26:
# Problem description

1. Built-In streaming plugins (UDP, ZMQ) not found (reason still unclear to me, a build on a different machine with same OS shows the two plugins)
2.  When PJ does not find a plugin (and the dropdown list seems therefore not to be populated), a click on the wheel beside the plugin dropdown (called "plugin specific options") crashes the program. Looks like 

`void MainWindow::on_buttonStreamingOptions_clicked()

{
  auto streamer = _data_streamer.at(ui->comboStreaming->currentText());
`

causes an exception in this case, because there is no element in the List

# Steps to reproduce
- compiled current main branch in debug configuration using cmake under Win10, 64bit
- use winqtdeploy (or manually) copy dependecies (.dlls to build dir)
- use gdb to stop on any exception
-  (debug build files are huge, which is why i did not upload my build)

facontidavide, 2021-12-29:
have you copied all the PlotJuggler plugins (DLL), alongside the executable?

Check the console message, you should see the list of loaded plugins.

Anyway, I should fix this, I don't want this to crash.

patzf, 2021-12-29:
Yes, the plugins are alongside the executable (as they are right after the build). But as I tried to mention above, It is a minor bug, since the plugins are recognised properly when the correct "version" of  libstdc++-6.dll is deployed, so it is really ONLY a problem If you do not compile any plugins and still click onto that button, should be rarely the case ;-)

btw: just checked 3.3.4: ZMQ plugin was kicked out in favor of MQTT? is it a replacement or just a new "default" of plugins that are compiled together with the build process of PJ?

facontidavide, 2021-12-29:
Zmq and mqtt should both be available, if you have the right dependencies include in your system. Let me double check on windows.
Dependencies are supported through vcpkg

patzf, 2021-12-29:
ok, thanks! I will give it a try with vcpkg

patzf, 2022-01-01:
No luck to install the dependencies so far with msys2/mingw64 until now. Even on a clean Win10 machine with fresh setup vcpkg fails to install mosquitto and protobuf - at least with MinGW. Guess I have to try the visualstudio compiler set..

commit fd708e8e (2022-01-29): fix #575
closed 2022-01-29 by facontidavide