PlotJuggler/PlotJuggler #576: PJ seems not plot well for high frequency data ~ 200Hz

extr15, 2021-12-28:
Thanks for your great tool, it helps a lot!

# Problem description

When play a rosbag and plot the timestamp of a 200Hz data, it seems PJ plots not right (the red one, I have another 50Hz data, the green one as a comparison). The timestamp increase 5ms for each data, thus the plot should be a line, while it not:
[image]
If I play rosbag with `-r 0.5`( play rate * 0.5), the plot seems better but still not right:
[image]
Instead if I load the rosbag to PJ directly and plot the timestamp, it is right(the two data coincide):
[image]
My system is `Linux 5.11.0-43-generic #47~20.04.2-Ubuntu`.
Thanks.

facontidavide, 2021-12-29:
My educated guess is that this is related to TCp transport https://en.wikipedia.org/wiki/Nagle%27s_algorithm

Unlerss you specify "use the timestamp from header.stamp", the timestamp used will be based on when PlotJuggler "received" the message and, because of the Nagle algorithm, messages at this frequency are sent in "batches".

closed 2021-12-29 by facontidavide

extr15, 2021-12-30:
@facontidavide , Thanks for the reply and it makes sense.
Does PJ support "use the timestamp from header.stamp"?