PlotJuggler/PlotJuggler #582: Memory leak?

hogAlert, 2022-01-07:
# Problem description

Memory consumed by the application increases each time new data/layout is loaded until eventually the system starts to choke due to lack of memory (can be as few as 10 reloads).

# Steps to reproduce (important)

In order to 'refresh' the plots when the source data has changes, I do the following sequence:
- Menu/App/Delete everything
- Layout/Load data and plots

However each time I do this, the memory consumed by the application increases (as view by system monitor e.g. htop "RES") until after a few (maybe 10) loops the system starts to choke due to lack of memory.

The memory consumed for each 'loop' as reported by htop "RES" is:

| step                                      | memory required |
| ----------------------------------------- | --------------- |
| app first open (no data loaded)           | 111M            |
| after layout loaded (and referenced data) | 223M            |
| after data clear                          | 223M            |
| reload 1                                  | 333M            |
| after data clear                          | 333M            |
| reload 2                                  | 441M            |
| after data clear / reload 3               | 551M            |
| after data clear / reload 4               | 660M            |
| after data clear / reload 5               | 769M            |
| ...                                       |                 |
| app close and re-open (no data loaded)    | 109M            |

I also tried using the 'clear data points' menu item and the trashcan icon/delete all or delete points (each seems to give the same result)

# Describe your platform / Operative System.
Manjaro Linux / Plotjuggler v3.3.5 via from AUR.
The problem was first noticed on 3.3.1, but I edited the AUR build file to pull and build 3.3.5 and the same issue persists in this version (latest available as of right now).

# If it can't be reproduced with the dummy data, __please__ share the CSV file or the rosbag that can be used to reproduce the problem.

I don't know what a rosbag is (sounds dirty) and I didn't try with the sample data (I can if you really need me to) but I attach my sample data and layout that shows the problem.
This data file is ~1.6megabytes (not that big?)

[plotjugglerMemoryLeakData.zip](https://github.com/facontidavide/PlotJuggler/files/7831632/plotjugglerMemoryLeakData.zip)

hogAlert, 2022-01-09:
Can also confirm same thing happens in windows (V3.3.0)
It's also not necessary to do the first step (Menu/App/Delete everything), just re-loading the layout (which has the data source stored) will consume ever increasing amounts of memory.

facontidavide, 2022-01-09:
I will look at this today. Thanks for reporting

facontidavide, 2022-01-09:
Are you compiling in release or debug mode?

hogAlert, 2022-01-09:
> Are you compiling in release or debug mode?

I'd guess release  - it'll be in the AUR build file somewhere I guess:
package info: https://aur.archlinux.org/packages/plotjuggler/
build file: https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=plotjuggler

the relevant part seems to be:

build() {
	cd PlotJuggler-"$pkgver"
	mkdir -p build
	cd build
	cmake ..
	make
}

facontidavide, 2022-01-10:
Solution pushed. Let me know if it solves your problems.

Thanks a lot for reporting

[image]

closed 2022-01-10 by facontidavide

hogAlert, 2022-01-10:
Thanks, I'll check it out asap.

In the meantine..

[image]

hogAlert, 2022-01-10:
I pulled and build the current master (b199cf5), not sure it's fixed..?
(note system memory shown filtered for PJ in the window above/behind)

[image]

facontidavide, 2022-01-10:
On my computer it seems it might be related to OpenGL. Can you disable OpenGL in **Preferences**, relaunch PlotJuggler and check if anything chnages?

hogAlert, 2022-01-10:
Yeah that's miles better.
Much less memory to start with, and although it goes up initially during the first few data reloads, it seems to settle out at around 65MB (compared with 500MB+ before).
Cheers.

facontidavide, 2022-01-10:
This really sucks. OpenGL makes visualization so useful, and I have no idea how to fix this :(

hogAlert, 2022-01-10:
I feel for you man, sucks when stuff you need to fix is out of your control. I suppose you could file a bug request.. :o
In the meantime, unless I'm missing something, it seems to work fine with it disabled.