PlotJuggler/PlotJuggler #583: ROS2 best-efforts topics do not work in PlotJuggler

bijoua29, 2022-01-20:
# Problem description
PlotJuggler does not work for ROS2 publishers of best-efforts topics.

# Steps to reproduce (important)
I am on running on Ubuntu 20.04 running the latest ros2 rolling from Jan 2022. In the Streaming-Start dialog for a ROS2 Topic Subscriber, I select a topic that I know is published as best-efforts. The topic does not show up in the 'Timeseries List' so I can't select it for plotting. I get the following warning at the shell where I started PlotJuggler:

[WARN] [1642649864.421922384] [plotjuggler]: New publisher discovered on topic '/athena/starboard/joint_error_fb', offering incompatible QoS. No messages will be sent to it. Last incompatible policy: RELIABILITY_QOS_POLICY

Unfortunately the rosbag topics in datasamples/rosbag2_test are all reliable so I can't reproduce this problem with the test rosbag. If there was a best-efforts topic in this test rosbag, I'm sure it would reproduce the problem. Also, I can't really attach a rosbag from my application since it is proprietary data.

# History
I believe the best-efforts functionality has been broken for a while. Our team switched to ros2 rolling sometime around Sep 2021 and I think the best-efforts has not worked since then for us in PlotJuggler. It was working for us prior to that when we were on ros2 foxy.

My guess is that this functionality broke when PlotJuggler switched to using the node->create_generic_subscription() API. It doesn't look like it passes a qos to node->create_generic_subscription() in https://github.com/PlotJuggler/plotjuggler-ros-plugins/blob/rolling/plugins/DataStreamROS2/datastream_ros2.cpp#L201, thereby defaulting to a reliable subscription. A reliable subscriber of a best-efforts publisher is incompatible.

When I look at the use of create_generic_subscription() in rosbag2 at https://github.com/ros2/rosbag2/blob/master/rosbag2_transport/src/rosbag2_transport/recorder.cpp#L255, it actually determines the lowest denominator qos from all the known publishers of the topic. So it seems PlotJuggler should be doing something similar.

DaniGarciaLopez, 2022-02-08:
We are also unable to visualize a topic with `best_effort` reliability. The topic appears in the list but after adding it, it's not showing in the timeseries list. We are using the latest binary in galactic (`ros-galactic-plotjuggler-ros:amd64/focal 1.5.1-1focal.20211223.041145 uptodate`) with Plotjuggler version 3.3.0

bijoua29, 2022-02-09:
> We are also unable to visualize a topic with `best_effort` reliability. The topic appears in the list but after adding it, it's not showing in the timeseries list. We are using the latest binary in [...]

Good. I thought it was just our team. It is an issue for us since we use best-efforts extensively and our team has stopped using PlotJuggler as a tool, for the time being.

closed 2022-02-11 by facontidavide

bijoua29, 2022-11-30:
And I am still getting this same error on the latest rolling sync from 2022-11-16. I have been compiling plotjuggler-ros-plugins from source which is at 1.7.1. However ros-rolling-plotjuggler-ros has only been released to ver 1.5.1. I'm not sure why the plotjuggler-ros hasn't been updated to 1.7.1. Till then, I guess I have to compile it from source but I shouldn't have to.

I'm not sure why this issue was closed. Till 1.7.1 version of plotjuggler-ros is not released, it should not be closed.