PlotJuggler/PlotJuggler #586: Plotjuggler Crashes Intermittently

Harry-Lukasz, 2022-01-26:
# Problem description
When loading data from different file types, PlotJuggler with intermittently crash. For example, loading a CSV file and ULOG file for comparisons causes intermittent crashes.

# System:
- Windows 10 Version 10.0.19043
- Intel Core i7 9th Gen
- 16GM RAM
# Steps to reproduce:
- Load ULOG file
- Load CSV file
- Attempt to drag various data attributes onto graph and eventually program crashes (after a few minutes)

facontidavide, 2022-01-27:
I need to release a new window binary, this might be solved already

facontidavide, 2022-02-11:
I f the problem persist with version 3.4.1, send me the files to replicate the error on my side and open this issue again

closed 2022-02-11 by facontidavide