PlotJuggler/PlotJuggler #587: I can only perform Quaternion to RPY once

sverrevr, 2022-01-28:
# Problem description
If I have performed tools/"Quaternion to RPY" once then I cannot do it again (on other data). The save button the second time is grayed out and cant be pressed. 

[image]

# Steps to reproduce (important)

- On windows 10
- Load some data
- Perform tools/"Quaternion to RPY" once
- Perform tools/"Quaternion to RPY" again

sverrevr, 2022-01-28:
Recreated on master on wsl2 running Ubuntu 18.04

facontidavide, 2022-01-28:
I guess you want to overwrite the old values, right?

sverrevr, 2022-01-28:
I wanted to define further values, but overwriting would also be useful. 
So even though I give it a new name I am unable to press save

facontidavide, 2022-01-29:
Just pushed a fix (and more). Check latest commit on **main**

closed 2022-01-29 by facontidavide

sverrevr, 2022-01-31:
Works like a charm! I'm impressed with how fast you solved this issue! thanks :)