PlotJuggler/PlotJuggler #588: Unable to load layout with custom seiers

sverrevr, 2022-01-28:
# Problem description
If I make a custom series and save the layout, then I cannot load it again. Attempting to load it again causes the program to crash:
[image]

# Steps to reproduce (important)

- On Windows 10 
- Make any custom series
- Save the layout
- Close and re-open plotjuggler 
- Load the layout

sverrevr, 2022-01-28:
I recreated the problem on wsl2-0 running Ubuntu 18.04 with the ROS installation of plotjuggler. 
The following error message was produced:
`WARNING: Application calling GLX 1.3 function "glXCreatePbuffer" when GLX 1.3 is not supported!  This is an application bug!
corrupted size vs. prev_size`

sverrevr, 2022-01-28:
The issue seems to be solved on master.
But once I got the following error when i:
- Loaded data
- Made on custom seiers, but did not move it to any plot
- Saved the layout
- Closed and re-opened plotjuggler 
- Loaded the layout
- Dragged the custom series to a plot

This error happened twice in a row, and then I could no longer reproduce it... 

`WARNING: Application calling GLX 1.3 function "glXCreatePbuffer" when GLX 1.3 is not supported!  This is an application bug!
Falling back to using screens root_visual.
Stack trace (most recent call last):
#31   Object "/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x7efefb3c15ea, in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*)
#30   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7efefadff8a7, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#29   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7efefbb85103, in QApplication::notify(QObject*, QEvent*)
#28   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7efefbb7d877, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#27   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7efefadff43b, in QCoreApplicationPrivate::sendThroughApplicationEventFilters(QObject*, QEvent*)
#26   Object "/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x7efefb3b46b8, in QBasicDrag::eventFilter(QObject*, QEvent*)
#25   Object "/usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5", at 0x7efef4373c16, in QXcbIntegration::createPlatformWindow(QWindow*) const
#24   Object "/usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5", at 0x7efef43716e6, in QXcbIntegration::createPlatformWindow(QWindow*) const
#23   Object "/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x7efefb399c5e, in QWindowSystemInterface::handleDrop(QWindow*, QMimeData const*, QPoint const&, QFlags<Qt::DropAction>)
#22   Object "/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x7efefb3c2a35, in QGuiApplicationPrivate::processDrop(QWindow*, QMimeData const*, QPoint const&, QFlags<Qt::DropAction>)
#21   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7efefadff8a7, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#20   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7efefbb85103, in QApplication::notify(QObject*, QEvent*)
#19   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7efefbb7d83b, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#18   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7efefbbd9809, in QDesktopWidget::qt_metacall(QMetaObject::Call, int, void**)
#17   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7efefbbd8aad, in QDesktopWidget::qt_metacall(QMetaObject::Call, int, void**)
#16   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7efefadff8a7, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#15   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7efefbb85272, in QApplication::notify(QObject*, QEvent*)
#14   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7efefbb7d83b, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#13   Source "/home/sverrevr/plotjuggler/PlotJuggler/3rdparty/qwt/src/qwt_plot.cpp", line 239, in QwtPlot::event(QEvent*) [0x55b5a25bcfe5]
        236:  */
        237: bool QwtPlot::event( QEvent* event )
        238: {
      > 239:     bool ok = QFrame::event( event );
        240:     switch ( event->type() )
        241:     {
        242:         case QEvent::LayoutRequest:
#12   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7efefbc5d44d, in QFrame::event(QEvent*)
#11   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7efefbbbc037, in QWidget::event(QEvent*)
#10 | Source "/home/sverrevr/plotjuggler/PlotJuggler/plotjuggler_base/src/plotwidget_base.cpp", line 145, in PJ::PlotWidgetBase::QwtPlotPimpl::dropEvent(QDropEvent*)
    |   143:   void dropEvent(QDropEvent* event) override
    |   144:   {
    | > 145:     event_callback(event);
    |   146:   }
    |   147: };
      Source "/usr/include/c++/7/bits/std_function.h", line 706, in PJ::PlotWidgetBase::QwtPlotPimpl::dropEvent(QDropEvent*) [0x55b5a25ae3e8]
        703:     {
        704:       if (_M_empty())
        705:    __throw_bad_function_call();
      > 706:       return _M_invoker(_M_functor, std::forward<_ArgTypes>(__args)...);
        707:     }
        708:
        709: #if __cpp_rtti
#9    Source "/home/sverrevr/plotjuggler/PlotJuggler/build/plotjuggler_base/include/PlotJuggler/moc_plotwidget_base.cpp", line 210, in PJ::PlotWidgetBase::dropSignal(QDropEvent*) [0x55b5a25a40bf]
        207: void PJ::PlotWidgetBase::dropSignal(QDropEvent * _t1)
        208: {
        209:     void *_a[] = { nullptr, const_cast<void*>(reinterpret_cast<const void*>(&_t1)) };
      > 210:     QMetaObject::activate(this, &staticMetaObject, 3, _a);
        211: }
        212:
        213: // SIGNAL 4
#8    Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7efefae2e63e, in QMetaObject::activate(QObject*, int, int, void**)
#7    Source "/home/sverrevr/plotjuggler/PlotJuggler/plotjuggler_app/plotwidget.cpp", line 507, in PlotWidget::onDropEvent(QDropEvent*) [0x55b5a24c2984]
        504:   if (curves_changed)
        505:   {
        506:     emit curvesDropped();
      > 507:     emit curveListChanged();
[... 44 lines cut ...]
        124: {
        125:     void *_a[] = { nullptr, const_cast<void*>(reinterpret_cast<const void*>(&_t1)) };
[... cut ...]
#0    Source "/home/sverrevr/plotjuggler/PlotJuggler/plotjuggler_app/plotwidget.cpp", line 910, in PlotWidget::on_changeTimeOffset(double) [0x55b5a24bbe70]
        907:   {
        908:     for (auto& it : curveList())
        909:     {
      > 910:       auto series = dynamic_cast<QwtSeriesWrapper*>(it.curve->data());
        911:       series->setTimeOffset(_time_offset);
        912:     }
        913:     if (!isXYPlot() && !curveList().empty())
Segmentation fault (Signal sent by the kernel [(nil)])
Segmentation fault`

facontidavide, 2022-01-29:
can not reproduce this in master. Can you give it a try?

sverrevr, 2022-01-31:
What I tried to say in the previous post is that I experienced a different error on master, with the error code in the previous post. But this error happened twice, and then never again. I am unable to reproduce it. 
So if the error code does not mean anything to you then I think we can close this issue.

facontidavide, 2022-01-31:
Try disable OpenGL in preferences

sverrevr, 2022-01-31:
I have opened a seperate issue with more infomration. #594

closed 2022-01-31 by sverrevr