PlotJuggler/PlotJuggler #594: Frequent crashing when dragging data into plotting area

sverrevr, 2022-01-31:
# Problem description

Quite often when I drag data into a plot plotjuggler crashes with error code `Segmentation fault (Address not mapped to object [0x20])` . See below for a detailed error message with reference to the code. 

The following might be relevant, or not:
- Running master on WSL2 running Ubuntu 18.04
- I can save the layout, drag and drop a time series that causes a crash, reload plotjuggler, drag and drop the same time series. The second time it works without a problem. 
- The error occurs quite often, sometimes every third time I drag a time-series (but it's not periodic at all). 
- I have only experienced it on custom series (but I have used a lot more custom series than normal ones)
- The series might be a bit complex looping over a window of past data and using trigonometric functions. 
- The error occurs with open GL disabled
- Im using ulg data
- The error message seems similar each time, seems to be related `dynamic_cast<QwtTimeseries*>`

I, unfortunately, cannot make the data I'm using publicly available, but I can share the functions I'm using if that is needed. 

```
Stack trace (most recent call last):
#31   Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7f50982c57fb, in g_main_context_iteration
#30   Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7f50982c576f, in g_main_context_dispatch
#29   Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7f50982c5536, in g_main_context_dispatch
#28   Object "/usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5", at 0x7f509451e25f, in QXcbGlIntegrationPlugin::qt_metacall(QMetaObject::Call, int, void**)
#27   Object "/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x7f509b4f133a, in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>)
#26   Object "/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x7f509b51a0b4, in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*)
#25   Object "/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x7f509b5185ea, in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*)
#24   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f509af568a7, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#23   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f509bcdc103, in QApplication::notify(QObject*, QEvent*)
#22   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f509bcd4877, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#21   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f509af5643b, in QCoreApplicationPrivate::sendThroughApplicationEventFilters(QObject*, QEvent*)
#20   Object "/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x7f509b50b6b8, in QBasicDrag::eventFilter(QObject*, QEvent*)
#19   Object "/usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5", at 0x7f50944cac16, in QXcbIntegration::createPlatformWindow(QWindow*) const
#18   Object "/usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5", at 0x7f50944c86e6, in QXcbIntegration::createPlatformWindow(QWindow*) const
#17   Object "/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x7f509b4f0c5e, in QWindowSystemInterface::handleDrop(QWindow*, QMimeData const*, QPoint const&, QFlags<Qt::DropAction>)
#16   Object "/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x7f509b519a35, in QGuiApplicationPrivate::processDrop(QWindow*, QMimeData const*, QPoint const&, QFlags<Qt::DropAction>)
#15   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f509af568a7, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#14   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f509bcdc103, in QApplication::notify(QObject*, QEvent*)
#13   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f509bcd483b, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#12   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f509bd30809, in QDesktopWidget::qt_metacall(QMetaObject::Call, int, void**)
#11   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f509bd2faad, in QDesktopWidget::qt_metacall(QMetaObject::Call, int, void**)
#10   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f509af568a7, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#9    Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f509bcdc272, in QApplication::notify(QObject*, QEvent*)
#8    Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f509bcd483b, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#7    Source "/home/sverrevr/plotjuggler/PlotJuggler/3rdparty/qwt/src/qwt_plot.cpp", line 239, in QwtPlot::event(QEvent*) [0x56002c9e00b5]
        236:  */
        237: bool QwtPlot::event( QEvent* event )
        238: {
      > 239:     bool ok = QFrame::event( event );
        240:     switch ( event->type() )
        241:     {
        242:         case QEvent::LayoutRequest:
#6    Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f509bdb444d, in QFrame::event(QEvent*)
#5    Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f509bd13037, in QWidget::event(QEvent*)
#4  | Source "/home/sverrevr/plotjuggler/PlotJuggler/plotjuggler_base/src/plotwidget_base.cpp", line 145, in PJ::PlotWidgetBase::QwtPlotPimpl::dropEvent(QDropEvent*)
    |   143:   void dropEvent(QDropEvent* event) override
    |   144:   {
    | > 145:     event_callback(event);
    |   146:   }
    |   147: };
      Source "/usr/include/c++/7/bits/std_function.h", line 706, in PJ::PlotWidgetBase::QwtPlotPimpl::dropEvent(QDropEvent*) [0x56002c9b21e8]
        703:     {
        704:       if (_M_empty())
        705:    __throw_bad_function_call();
      > 706:       return _M_invoker(_M_functor, std::forward<_ArgTypes>(__args)...);
        707:     }
        708:
        709: #if __cpp_rtti
#3    Source "/home/sverrevr/plotjuggler/PlotJuggler/build/plotjuggler_base/include/PlotJuggler/moc_plotwidget_base.cpp", line 210, in PJ::PlotWidgetBase::dropSignal(QDropEvent*) [0x56002c9a792f]
        207: void PJ::PlotWidgetBase::dropSignal(QDropEvent * _t1)
        208: {
        209:     void *_a[] = { nullptr, const_cast<void*>(reinterpret_cast<const void*>(&_t1)) };
      > 210:     QMetaObject::activate(this, &staticMetaObject, 3, _a);
        211: }
        212:
        213: // SIGNAL 4
#2    Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f509af8563e, in QMetaObject::activate(QObject*, int, int, void**)
#1    Source "/home/sverrevr/plotjuggler/PlotJuggler/plotjuggler_app/plotwidget.cpp", line 536, in PlotWidget::onDropEvent(QDropEvent*) [0x56002c8c5f0d]
        534:     for (const auto& curve_name : _dragging.curves)
        535:     {
      > 536:       bool added = addCurve(curve_name.toStdString()) != nullptr;
        537:       curves_changed = curves_changed || added;
        538:     }
        539:   }
#0    Source "/home/sverrevr/plotjuggler/PlotJuggler/plotjuggler_app/plotwidget.cpp", line 373, in PlotWidget::addCurve(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, QColor) [0x56002c8c1c4d]
        370:     info = PlotWidgetBase::addCurve(name, it2->second, color);
        371:   }
        372:
      > 373:   if( auto timeseries = dynamic_cast<QwtTimeseries*>( info->curve->data() ))
        374:   {
        375:     timeseries->setTimeOffset(_time_offset);
        376:   }
Segmentation fault (Address not mapped to object [0x20])
Segmentation fault
```

facontidavide, 2022-02-11:
Not sure why this is happening, but I pushed something that should at least fix the crash

sverrevr, 2022-02-11:
Thanks :) Ill test it out and report back 👍

brkay54, 2022-02-11:
Hi, I also have same issue. I tested the main after issue closed but it is not solved.

```
Loading compatible plugins from directory:  "/home/berkay/ws_plotjuggler/install/plotjuggler_ros/lib/plotjuggler_ros"
"libDataLoadROS2.so is a DataLoader plugin"
"libDataStreamROS2.so is a DataStreamer plugin"
[WARN] [1644571829.314527901] [rclcpp]: logging was initialized more than once
"libTopicPublisherROS2.so is a StatePublisher plugin"
Number of plugins loaded:  3 

Loading compatible plugins from directory:  ""
Number of plugins loaded:  0 

Loading compatible plugins from directory:  "/home/berkay/ws_plotjuggler/install/plotjuggler/lib/plotjuggler"
"libDataLoadCSV.so is a DataLoader plugin"
"libDataLoadULog.so is a DataLoader plugin"
"libDataStreamSample.so is a DataStreamer plugin"  ...disabled, unless option -t is used
"libDataStreamUDP.so is a DataStreamer plugin"
"libDataStreamWebSocket.so is a DataStreamer plugin"
"libProtobufParser.so is a MessageParser plugin"
"libPublisherCSV.so is a StatePublisher plugin"
Type conversion already registered from type QString to type QwtText
"libToolboxFFT.so is a Toolbox plugin"
"libToolboxLuaEditor.so is a Toolbox plugin"
Type conversion already registered from type QString to type QwtText
"libToolboxQuaternion.so is a Toolbox plugin"
Number of plugins loaded:  9 

Loading compatible plugins from directory:  "/home/berkay/.local/share/PlotJuggler"
Number of plugins loaded:  0 

Stack trace (most recent call last):
#31   Object "/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x7f75634757d2, in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*)
#30   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f756308a809, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#29   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f7563aae0ef, in QApplication::notify(QObject*, QEvent*)
#28   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f7563aa4a9f, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#27   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f756308a3ea, in QCoreApplicationPrivate::sendThroughApplicationEventFilters(QObject*, QEvent*)
#26   Object "/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x7f75634af22b, in QBasicDrag::eventFilter(QObject*, QEvent*)
#25   Object "/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5", at 0x7f755e773535, in QXcbConnection::initializeScreens()
#24   Object "/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5", at 0x7f755e770118, in QXcbConnection::initializeScreens()
#23   Object "/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x7f7563450bcf, in QWindowSystemInterface::handleDrop(QWindow*, QMimeData const*, QPoint const&, QFlags<Qt::DropAction>, QFlags<Qt::MouseButton>, QFlags<Qt::KeyboardModifier>)
#22   Object "/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x7f75634775b1, in QGuiApplicationPrivate::processDrop(QWindow*, QMimeData const*, QPoint const&, QFlags<Qt::DropAction>, QFlags<Qt::MouseButton>, QFlags<Qt::KeyboardModifier>)
#21   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f756308a809, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#20   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f7563aae0ef, in QApplication::notify(QObject*, QEvent*)
#19   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f7563aa4a65, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#18   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f7563b06429, in QDesktopWidget::qt_metacall(QMetaObject::Call, int, void**)
#17   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f7563b05691, in QDesktopWidget::qt_metacall(QMetaObject::Call, int, void**)
#16   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f756308a809, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#15   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f7563aae7a9, in QApplication::notify(QObject*, QEvent*)
#14   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f7563aa4a65, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#13   Source "/home/berkay/ws_plotjuggler/src/PlotJuggler/3rdparty/qwt/src/qwt_plot.cpp", line 239, in QwtPlot::event(QEvent*) [0x562cdfdf774b]
        236:  */
        237: bool QwtPlot::event( QEvent* event )
        238: {
      > 239:     bool ok = QFrame::event( event );
        240:     switch ( event->type() )
        241:     {
        242:         case QEvent::LayoutRequest:
#12   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f7563b94d51, in QFrame::event(QEvent*)
#11   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f7563ae72b5, in QWidget::event(QEvent*)
#10 | Source "/home/berkay/ws_plotjuggler/src/PlotJuggler/plotjuggler_base/src/plotwidget_base.cpp", line 146, in PJ::PlotWidgetBase::QwtPlotPimpl::dropEvent(QDropEvent*)
    |   144:   void dropEvent(QDropEvent* event) override
    |   145:   {
    | > 146:     event_callback(event);
    |   147:   }
    |   148: };
      Source "/usr/include/c++/9/bits/std_function.h", line 688, in std::function<void (QEvent*)>::operator()(QEvent*) const [0x562cdfdc2c1c]
        685:     {
        686:       if (_M_empty())
        687: 	__throw_bad_function_call();
[... 75 lines cut ...]
        129:     void *_a[] = { nullptr, const_cast<void*>(reinterpret_cast<const void*>(&_t1)) };
      > 130:     QMetaObject::activate(this, &staticMetaObject, 0, _a);
        131: }
        132: QT_WARNING_POP
        133: QT_END_MOC_NAMESPACE
[... cut ...]
        242:         case 26: _t->onFlipAxis(); break;
#0    Source "/home/berkay/ws_plotjuggler/src/PlotJuggler/plotjuggler_app/plotwidget.cpp", line 996, in PlotWidget::on_changeTimeOffset(double) [0x562cdfcd8fc0]
        993:   {
        994:     for (auto& it : curveList())
        995:     {
      > 996:       auto series = dynamic_cast<QwtTimeseries*>(it.curve->data());
        997:       if( series )
        998:       {
        999:         series->setTimeOffset(_time_offset);
Segmentation fault (Signal sent by the kernel [(nil)])

```

facontidavide, 2022-02-12:
this is killing me! Working on it until I find this

facontidavide, 2022-02-12:
ok, I found the origin of multiple crashes, finally. Please check version 3.4.2

commit 9825a49d (2022-02-11): Fix #594
closed 2022-02-12 by facontidavide