PlotJuggler/PlotJuggler #609: Compiling with ubuntu 18.04 fatal error: QScopeGuard: No such file or directory

cho934, 2022-02-16:
Hello,

# Problem description

I'm trying to build the last github version and got an fatal error: QScopeGuard: No such file or directory in ubuntu 18.04.
I use the qt5-default:

```
sudo apt-get install qt5-default
Reading package lists... Done
Building dependency tree       
Reading state information... Done
qt5-default is already the newest version (5.9.5+dfsg-0ubuntu2.6).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
````

```
...
[ 75%] Generating ui_integral_transform.h
Scanning dependencies of target plotjuggler
[ 75%] Building CXX object plotjuggler_app/CMakeFiles/plotjuggler.dir/cheatsheet/cheatsheet_dialog.cpp.o
[ 76%] Building CXX object plotjuggler_app/CMakeFiles/plotjuggler.dir/customtracker.cpp.o
[ 76%] Building CXX object plotjuggler_app/CMakeFiles/plotjuggler.dir/curvelist_panel.cpp.o
[ 76%] Building CXX object plotjuggler_app/CMakeFiles/plotjuggler.dir/curvelist_view.cpp.o
[ 76%] Building CXX object plotjuggler_app/CMakeFiles/plotjuggler.dir/curvetree_view.cpp.o
[ 77%] Building CXX object plotjuggler_app/CMakeFiles/plotjuggler.dir/dummy_data.cpp.o
[ 77%] Building CXX object plotjuggler_app/CMakeFiles/plotjuggler.dir/main.cpp.o
/home/pmx/git/PlotJuggler/plotjuggler_app/main.cpp:14:10: fatal error: QScopeGuard: No such file or directory
 #include <QScopeGuard>
          ^~~~~~~~~~~~~
compilation terminated.
plotjuggler_app/CMakeFiles/plotjuggler.dir/build.make:577: recipe for target 'plotjuggler_app/CMakeFiles/plotjuggler.dir/main.cpp.o' failed
make[2]: *** [plotjuggler_app/CMakeFiles/plotjuggler.dir/main.cpp.o] Error 1
CMakeFiles/Makefile2:875: recipe for target 'plotjuggler_app/CMakeFiles/plotjuggler.dir/all' failed
make[1]: *** [plotjuggler_app/CMakeFiles/plotjuggler.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```

# Steps to reproduce (important)
```
cmake ..
make
```

QScopeGuard is more recent, and seems now to be mandatory, so it need Qt5.12 minimum ?

BR,

facontidavide, 2022-02-17:
That was recently added and it is not mandatory. I can remove it

closed 2022-02-17 by facontidavide

cho934, 2022-02-17:
It still need here.
```
[ 77%] Building CXX object plotjuggler_app/CMakeFiles/plotjuggler.dir/main.cpp.o
/home/pmx/git/PlotJuggler/plotjuggler_app/main.cpp: In function ‘int main(int, char**)’:
/home/pmx/git/PlotJuggler/plotjuggler_app/main.cpp:345:24: error: ‘qScopeGuard’ was not declared in this scope
     auto sentryClose = qScopeGuard([] { sentry_close(); });
                        ^~~~~~~~~~~
plotjuggler_app/CMakeFiles/plotjuggler.dir/build.make:577: recipe for target 'plotjuggler_app/CMakeFiles/plotjuggler.dir/main.cpp.o' failed
```

facontidavide, 2022-02-18:
Yes, need to find time to fix this

commit 6f513c07 (2022-02-17): fix #609