PlotJuggler/PlotJuggler #612: Segmentation fault when saving axis settings

ClemensVolk, 2022-02-24:
# Problem description
When saving axis values for plots, the program crashes with a segmentation fault.

# Steps to reproduce

- [Ubuntu 20.04.3 LTS (Focal Fossa)]
- ROS 2 Foxy Fitzroy

Subscribe to a ros2 topic and change the y-axis to take on fixed values instead of changing with the incoming data points. Right click on plot->first option->right bottom corner of the dialog. Select "Max" and "Min". Either type in own values or use the provided button for  setting default ones. Then save the changes. It then segfaults with the following output:

```
Stack trace (most recent call last):
#31   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f3daf657434, in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)
#30   Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7f3dae5ce4a2, in g_main_context_iteration
#29   Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7f3dae5ce3ff, in g_main_context_dispatch
#28   Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7f3dae5ce17c, in g_main_context_dispatch
#27   Object "/usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5", at 0x7f3dab7e232d, in QXcbNativeInterface::dumpNativeWindows(unsigned long long) const
#26   Object "/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x7f3daf9c435a, in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>)
#25   Object "/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x7f3daf9ea10a, in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*)
#24   Object "/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x7f3daf9e87d2, in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*)
#23   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f3daf5ff809, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#22   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f3db00210ef, in QApplication::notify(QObject*, QEvent*)
#21   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f3db0017a65, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#20   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f3db00791eb, in QDesktopWidget::qt_metacall(QMetaObject::Call, int, void**)
#19   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f3db007635c, in QDesktopWidget::qt_metacall(QMetaObject::Call, int, void**)
#18   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f3db0020456, in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool)
#17   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f3daf5ff809, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#16   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f3db0021342, in QApplication::notify(QObject*, QEvent*)
#15   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f3db0017a65, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#14   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f3db005a2d0, in QWidget::event(QEvent*)
#13   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f3db010d254, in QAbstractButton::mousePressEvent(QMouseEvent*)
#12   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f3db010cb00, in QAbstractButton::toggled(bool)
#11   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f3daf62b28b, in QMetaObject::activate(QObject*, int, int, void**)
#10   Object "/opt/ros/foxy/lib/plotjuggler/plotjuggler", at 0x55dccf3c055f, in 
#9    Object "/opt/ros/foxy/lib/plotjuggler/plotjuggler", at 0x55dccf3bdc0c, in 
#8    Object "/opt/ros/foxy/lib/plotjuggler/plotjuggler", at 0x55dccf42da64, in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > std::operator+<char, std::char_traits<char>, std::allocator<char> >(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
#7    Object "/opt/ros/foxy/lib/plotjuggler/plotjuggler", at 0x55dccf42b0f1, in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > std::operator+<char, std::char_traits<char>, std::allocator<char> >(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
#6    Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f3daf62b327, in QMetaObject::activate(QObject*, int, int, void**)
#5    Object "/opt/ros/foxy/lib/plotjuggler/plotjuggler", at 0x55dccf3e25e2, in void std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_construct<char const*>(char const*, char const*, std::forward_iterator_tag)
#4    Object "/opt/ros/foxy/lib/plotjuggler/plotjuggler", at 0x55dccf3e25ad, in void std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_construct<char const*>(char const*, char const*, std::forward_iterator_tag)
#3    Object "/opt/ros/foxy/lib/plotjuggler/plotjuggler", at 0x55dccf3bd981, in 
#2    Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f3daf62b1cf, in QMetaObject::activate(QObject*, int, int, void**)
#1    Object "/opt/ros/foxy/lib/plotjuggler/plotjuggler", at 0x55dccf3be39c, in 
#0    Object "/opt/ros/foxy/lib/plotjuggler/plotjuggler", at 0x55dccf422780, in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > std::operator+<char, std::char_traits<char>, std::allocator<char> >(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
Segmentation fault (Address not mapped to object [0x30])
```

facontidavide, 2022-03-06:
I am not able to replicate this on master. It could be that the problem has been fixed already. 
Which version of PlotJuggler are you using?

facontidavide, 2022-07-05:
closing for inactivity. If you can reproduce on **main** branch, let me know

closed 2022-07-05 by facontidavide