PlotJuggler/PlotJuggler #613: Possible issue with ROS 1 "sensor_msgs/JointState" - array fields not displayed

marco-tranzatto, 2022-03-01:
First of all, let me thank you for creating and making this awesome software available!

# Problem description

ROS1 message of type "sensor_msgs/JointState" can be echoed on terminal but its array fields do not show up when trying to plot them (after rosbag has been loaded into plotjuggler). I also tried the solution from issue #374 but it did not help. See below screenshot.

[image]

# Steps to reproduce (important)

- **Describe your platform / Operative System:** Ubuntu 20.04, ROS noetic, plotjuggler 3.4.1 (as shipped with `apt install ros-noetic-plotjuggler`)
- **Check if the problem can be reproduced using the dummy data created by the command line argument "-t" or one of the files in the folder "datasamples"**: sample data containing array can be correctly visualized.
- **If it can't be reproduced with the dummy data, __please__ share the CSV file or the rosbag that can be used to reproduce the problem.**: rosbag with example topic - [link](https://drive.google.com/file/d/11lo854ItYTXzG0KP4cBs-YZtsV0dvC6b/view?usp=sharing)

Thanks for any hint you might have on the issue.
Marco

facontidavide, 2022-03-01:
This is happening because your ** name** array is empty.
This case is not taken into account

marco-tranzatto, 2022-03-01:
thanks for your swift reply and explanation @facontidavide ! I'll update my code to make sure this does not happen.

closed 2022-03-01 by marco-tranzatto