PlotJuggler/PlotJuggler #617: I download the 2.4.0 of windows and install it. But when I open it , the software flashback

khadegar, 2022-03-10:
Thanks for contributing to PlotJuggler. You are great!

# Problem description

Describe the problem and the expected behavior. Use screenshots of video if approriate.

# Steps to reproduce (important)

- Describe your platform / Operative System.
- Check if the problem can be reproduced using the dummy data created by the command line argument "-t" or one of the files in the folder "datasamples".
- If it can't be reproduced with the dummy data, __please__ share the CSV file or the rosbag that can be used to reproduce the problem.

facontidavide, 2022-03-10:
That text is my template. Aren't you taking the time to describe what "flashback" means? Seriously?

khadegar, 2022-03-10:
I am very sorry.  I mean when I open it, it close after about 2 seconds.  I download the file  at your link
I am very thank you for your help

facontidavide, 2022-03-10:
I will submit soon 3.4.3. Let me know if problem persists

facontidavide, 2022-05-14:
close for inactivity

closed 2022-05-14 by facontidavide