PlotJuggler/PlotJuggler #618: Installation via apt for ROS galactic not working

topalbu, 2022-03-17:
Could be related to #351 

I'm running on Ubuntu focal (20.04.4 LTS) 

Only the plotjuggler-ros not listed in the source list for [galactic](http://packages.ros.org/ros2/ubuntu/dists/focal/main/binary-amd64/Packages). The other packages can be installed.

ajay1606, 2022-03-19:
Hello, i am also facing same issues in installing in ROS2 galactic distribution ! Any fix available ?

Appreciate any response !

TitouanTyack, 2022-03-19:
I had the same issue, so I built it from source and it worked. I followed the installation from this repo : https://github.com/PlotJuggler/plotjuggler-ros-plugins . Almost all packages have a galactic branch so I used git checkout ( I think you can do also git clone -b ). The colcon build passed with some warnings but plotjuggler-ros is working for me. (You need to use ros2 run and not launch as written in the readme)

ajay1606, 2022-03-19:
@TitouanTyack , Thank you so much for quick response, its working well as you mentioned it. Really appreciate your help :)

msadowski, 2022-03-23:
In case anyone needs to move fast, this is what I've run following the plothuggler-ros-plugins instructions:

```
 mkdir -p ~/ws_plotjuggler/src
 cd ~/ws_plotjuggler/src
 git clone https://github.com/PlotJuggler/plotjuggler_msgs.git #f6055cd8896d
 git clone https://github.com/facontidavide/PlotJuggler.git #cd8f994df
 git clone https://github.com/PlotJuggler/plotjuggler-ros-plugins.git -b galactic #bdf82a833569
 
 cd ~/ws_plotjuggler
rosdep install --from-paths src --ignore-src --rosdistro=galactic -y
sudo apt-get install ros-galactic-rosbag2-transport

colcon build
source install/setup.bash

ros2 run plotjuggler plotjuggler
```

connoranderson, 2022-03-23:
Thanks @msadowski !

facontidavide, 2022-05-14:
it now works

closed 2022-05-14 by facontidavide