PlotJuggler/PlotJuggler #624: Clicking "apply filter" after right clicking on a plot causes a core dump

NuclearManD, 2022-03-31:
Thanks for contributing to PlotJuggler. You are great!

# Problem description

Clicking "apply filter" after right clicking on a plot causes a core dump.  It might be notable that I was applying a filter to a function I defined inside plotJuggler, not to the original data itself.

# Steps to reproduce

1. Open your data
2. Make this custom function:
    ```lua
    v_last = 16.8
    I_last = 1
    
    function calc(time, value, v1) -- Note the v1 - we need two inputs to this function
        R = (value - v_last) / (I_last - v1)
        v_last = value
        I_last = v1
        
        return R
    end
    ```
3. Plot the function
4. Remove your plot, then plot your function again.  This is so that you don't have the base `value` plot in your chart
5. You should have a plot with ONLY the function in it, just one line.  No others.
6. Right click and select `apply filter to data`

- I'm on Linux Mint. `uname -a`: `Linux forcyte-xenon 5.11.0-34-generic #36~20.04.1-Ubuntu SMP Fri Aug 27 08:06:32 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux`
- Check if the problem can be reproduced using the dummy data created by the command line argument "-t" or one of the files in the folder "datasamples".
    **Yes, it reproduces with** `./PlotJuggler-3.4.1-x86_64.AppImage -t`

# Terminal log (I launch plotJuggler from the terminal)

```bash
nuclaer@forcyte-xenon:~/perm$ ./PlotJuggler-3.4.1-x86_64.AppImage 
Loading compatible plugins from directory:  "/tmp/.mount_PlotJuuFcsmD/usr/bin"
"libDataLoadCSV.so is a DataLoader plugin"
"libDataLoadULog.so is a DataLoader plugin"
"libDataStreamMQTT_Mosquitto.so is a DataStreamer plugin"
"libDataStreamSample.so is a DataStreamer plugin"  ...disabled, unless option -t is used
"libDataStreamUDP.so is a DataStreamer plugin"
"libDataStreamWebSocket.so is a DataStreamer plugin"
"libProtobufParser.so is a MessageParser plugin"
"libPublisherCSV.so is a StatePublisher plugin"
QtAV 1.13.0(Apr 10 2020, 19:10:20)
Multimedia framework base on Qt and FFmpeg.
Distributed under the terms of LGPLv2.1 or later.
Shanghai, China Copyright (C) 2012-2019 Wang Bin (aka. Lucas Wang) wbsecg1@gmail.com
Donate: http://qtav.org/donate.html
Source: https://github.com/wang-bin/QtAV
Home page: http://qtav.org
"libPublisherVideoViewer.so is a StatePublisher plugin"
Type conversion already registered from type QString to type QwtText
"libToolboxFFT.so is a Toolbox plugin"
"libToolboxLuaEditor.so is a Toolbox plugin"
Type conversion already registered from type QString to type QwtText
"libToolboxQuaternion.so is a Toolbox plugin"
Number of plugins loaded:  11 

Loading compatible plugins from directory:  "/home/nuclaer/.local/share/PlotJuggler"
Number of plugins loaded:  0 

Stack trace (most recent call last):
#31   Object "/tmp/.mount_PlotJuuFcsmD/usr/lib/libQt5Gui.so.5", at 0x7fab7225035a, in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>)
#30   Object "/tmp/.mount_PlotJuuFcsmD/usr/lib/libQt5Gui.so.5", at 0x7fab7227610a, in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*)
#29   Object "/tmp/.mount_PlotJuuFcsmD/usr/lib/libQt5Gui.so.5", at 0x7fab722747d2, in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*)
#28   Object "/tmp/.mount_PlotJuuFcsmD/usr/lib/libQt5Core.so.5", at 0x7fab71e52809, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#27   Object "/tmp/.mount_PlotJuuFcsmD/usr/lib/libQt5Widgets.so.5", at 0x7fab7290c0ef, in QApplication::notify(QObject*, QEvent*)
#26   Object "/tmp/.mount_PlotJuuFcsmD/usr/lib/libQt5Widgets.so.5", at 0x7fab72902a65, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#25   Object "/tmp/.mount_PlotJuuFcsmD/usr/lib/libQt5Widgets.so.5", at 0x7fab729641eb, in 
#24   Object "/tmp/.mount_PlotJuuFcsmD/usr/lib/libQt5Widgets.so.5", at 0x7fab72961ce3, in 
#23   Object "/tmp/.mount_PlotJuuFcsmD/usr/lib/libQt5Widgets.so.5", at 0x7fab7290b456, in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool)
#22   Object "/tmp/.mount_PlotJuuFcsmD/usr/lib/libQt5Core.so.5", at 0x7fab71e52809, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#21   Object "/tmp/.mount_PlotJuuFcsmD/usr/lib/libQt5Widgets.so.5", at 0x7fab7290c342, in QApplication::notify(QObject*, QEvent*)
#20   Object "/tmp/.mount_PlotJuuFcsmD/usr/lib/libQt5Widgets.so.5", at 0x7fab72902a65, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#19   Object "/tmp/.mount_PlotJuuFcsmD/usr/lib/libQt5Widgets.so.5", at 0x7fab72a8fada, in QMenu::event(QEvent*)
#18   Object "/tmp/.mount_PlotJuuFcsmD/usr/lib/libQt5Widgets.so.5", at 0x7fab729452b5, in QWidget::event(QEvent*)
#17   Object "/tmp/.mount_PlotJuuFcsmD/usr/lib/libQt5Widgets.so.5", at 0x7fab72a8d4d1, in QMenu::mouseReleaseEvent(QMouseEvent*)
#16   Object "/tmp/.mount_PlotJuuFcsmD/usr/lib/libQt5Widgets.so.5", at 0x7fab72a8c4ad, in 
#15   Object "/tmp/.mount_PlotJuuFcsmD/usr/lib/libQt5Widgets.so.5", at 0x7fab72a84d11, in 
#14   Object "/tmp/.mount_PlotJuuFcsmD/usr/lib/libQt5Widgets.so.5", at 0x7fab728feaa1, in QAction::activate(QAction::ActionEvent)
#13   Object "/tmp/.mount_PlotJuuFcsmD/usr/lib/libQt5Widgets.so.5", at 0x7fab728fc3e5, in QAction::triggered(bool)
#12   Object "/tmp/.mount_PlotJuuFcsmD/usr/lib/libQt5Core.so.5", at 0x7fab71e7e327, in QMetaObject::activate(QObject*, int, int, void**)
#11 | Source "/usr/include/x86_64-linux-gnu/qt5/QtCore/qobjectdefs_impl.h", line 439, in call<QtPrivate::List<>, void>
    | Source "/usr/include/x86_64-linux-gnu/qt5/QtCore/qobjectdefs_impl.h", line 256, in call
    | Source "/usr/include/x86_64-linux-gnu/qt5/QtCore/qobjectdefs_impl.h", line 146, in operator()
      Source "/home/dfaconti/tmp_pj/PlotJuggler/plotjuggler_app/plotwidget.cpp", line 147, in impl [0x55c063a0e606]
#10 | Source "/home/dfaconti/tmp_pj/PlotJuggler/plotjuggler_app/plotwidget_transforms.cpp", line 51, in setSelected
      Source "/usr/include/x86_64-linux-gnu/qt5/QtWidgets/qlistwidget.h", line 335, in DialogTransformEditor [0x55c063a1c4bd]
#9    Object "/tmp/.mount_PlotJuuFcsmD/usr/lib/libQt5Widgets.so.5", at 0x7fab72bb7805, in QListWidget::setItemSelected(QListWidgetItem const*, bool)
#8    Object "/tmp/.mount_PlotJuuFcsmD/usr/lib/libQt5Core.so.5", at 0x7fab71e100c0, in QItemSelectionModel::select(QModelIndex const&, QFlags<QItemSelectionModel::SelectionFlag>)
#7    Object "/tmp/.mount_PlotJuuFcsmD/usr/lib/libQt5Core.so.5", at 0x7fab71e15e96, in QItemSelectionModel::select(QItemSelection const&, QFlags<QItemSelectionModel::SelectionFlag>)
#6    Object "/tmp/.mount_PlotJuuFcsmD/usr/lib/libQt5Core.so.5", at 0x7fab71e132c1, in QItemSelectionModel::emitSelectionChanged(QItemSelection const&, QItemSelection const&)
#5    Object "/tmp/.mount_PlotJuuFcsmD/usr/lib/libQt5Core.so.5", at 0x7fab71e0de83, in QItemSelectionModel::selectionChanged(QItemSelection const&, QItemSelection const&)
#4    Object "/tmp/.mount_PlotJuuFcsmD/usr/lib/libQt5Core.so.5", at 0x7fab71e7e1cf, in QMetaObject::activate(QObject*, int, int, void**)
#3    Object "/tmp/.mount_PlotJuuFcsmD/usr/lib/libQt5Core.so.5", at 0x7fab71e7e28b, in QMetaObject::activate(QObject*, int, int, void**)
#2  | Source "/home/dfaconti/tmp_pj/build/plotjuggler_app/plotjuggler_autogen/EWIEGA46WW/moc_plotwidget_transforms.cpp", line 129, in qt_static_metacall
    | Source "/home/dfaconti/tmp_pj/build/plotjuggler_app/plotjuggler_autogen/EWIEGA46WW/moc_plotwidget_transforms.cpp", line 82, in qt_static_metacall
      Source "/home/dfaconti/tmp_pj/build/plotjuggler_app/plotjuggler_autogen/EWIEGA46WW/moc_plotwidget_transforms.cpp", line 88, in qt_metacall [0x55c0639a5246]
#1    Source "/home/dfaconti/tmp_pj/PlotJuggler/plotjuggler_app/plotwidget_transforms.cpp", line 121, in on_listCurves_itemSelectionChanged [0x55c063a1bb48]
#0  | Source "/home/dfaconti/tmp_pj/PlotJuggler/plotjuggler_base/src/timeseries_qwt.cpp", line 75, in shared_ptr
    | Source "/usr/include/c++/9/bits/shared_ptr.h", line 129, in __shared_ptr
    |   127:       constexpr shared_ptr() noexcept : __shared_ptr<_Tp>() { }
    |   128: 
    | > 129:       shared_ptr(const shared_ptr&) noexcept = default;
    |   130: 
    |   131:       /**
      Source "/usr/include/c++/9/bits/shared_ptr_base.h", line 1167, in transform [0x55c063b19b84]
       1164: 	: _M_ptr(__p), _M_refcount(__r._M_refcount) // never throws
       1165: 	{ }
       1166: 
      >1167:       __shared_ptr(const __shared_ptr&) noexcept = default;
       1168:       __shared_ptr& operator=(const __shared_ptr&) noexcept = default;
       1169:       ~__shared_ptr() = default;
Segmentation fault (Address not mapped to object [0x140])
Segmentation fault (core dumped)
```

AlexanderHauf, 2022-04-07:
I have the same problem, while viewing rosbag data and then clicking "apply filter".
There is no problem up to version 3.3.5. This problem only occurs from version 3.4.0 on.

facontidavide, 2022-04-07:
Already fixed in recent versions

closed 2022-04-07 by facontidavide