PlotJuggler/PlotJuggler #628: support for lcm data stream

binfengxueqian, 2022-04-20:
Thanks for contributing to PlotJuggler. You are great!

# Problem description

I am using the LCM library for data communication and hope the plothuggler to suppor LCM data stream and parse of LCM message formats

the github of LCM is https://github.com/lcm-proj/lcm

facontidavide, 2022-04-20:
Doable, I guess, but only as a paid development.
If anybody wants to sponsor this effort, let me know.

binfengxueqian, 2022-04-20:
> Doable, I guess, but only as a paid development. If anybody wants to sponsor this effort, let me know.

How much will it cost？

facontidavide, 2022-04-20:
Please reach me on davide.faconti@gmail.com 😊

closed 2022-08-10 by facontidavide

hxrkpib, 2023-10-01:
> Please reach me on [davide.faconti@gmail.com](mailto:davide.faconti@gmail.com) 😊

I also want this function and have sent you an email.

facontidavide, 2023-10-01:
I took a quick look at LCM and I don't find something mandatory in this case, to implement it with PlotJuggler, a reflection API for the serialized messages.

In other words, using this as a reference: http://lcm-proj.github.io/lcm/content/tutorial-cpp.html
I need to be able to subscribe WITHOUT including the type

`exlcm::example_t`

If you find an example that shows how to do that. Let me know.

lgkimjy, 2024-06-12:
Are there any updates on LCM support??