PlotJuggler/PlotJuggler #630: The time tracker is not appearing.

MoreTore, 2022-04-25:
Thanks for contributing to PlotJuggler. You are great!

The time tracker is not appearing. I am using this fork https://github.com/MoreTore/PlotJuggler compiled on ubuntu 22.04 using the dockerfile included in the fork. I have my binaries as a release if that helps.

Describe the problem and the expected behavior. Use screenshots of video if approriate.

# Steps to reproduce (important)
git clone https://github.com/MoreTore/PlotJuggler
cd PlotJuggler
git submodule update --init
./build.sh

I run this with openpilot/tools/replay

[image]

MoreTore, 2022-04-25:
If its an easy fix great, if not then its no big deal

facontidavide, 2022-05-14:
that is **intentional**. It is disabled when streaming is active and will be activated when streaming is stopped or paused.

closed 2022-05-14 by facontidavide