PlotJuggler/PlotJuggler #633: Test suite for the project

erickisos, 2022-05-01:
## Context

This is not a problem itself, but I realized that the project doesn't include any test suite, so I just want to know if it is on plans or if you think that it would be valuable to check how to add it incrementally, to simplify the development process (avoid breaking changes without explicitly getting failures on tests).

If so, maybe I can help with a basic implementation, but I want to know if it is valuable or not at the time.

facontidavide, 2022-05-14:
I would love to but I have zero time to dedicate to this.

I did manual Qt test in the past (Groot), but it is time consuming.
I wonder if this could be a solution: https://www.froglogic.com/squish/editions/qt-gui-test-automation/

If you want to help, I would appreciate that.

erickisos, 2022-05-22:
Maybe we can start with something simple, like adding unit tests just to logic functions, I will try by adding some minimal examples and hopefully we could gradually increase the coverage.
Thanks for your answer btw.

closed 2026-01-06 by facontidavide