PlotJuggler/PlotJuggler #640: NaN in JSON is not parsed correctly

pavel-kirienko, 2022-05-07:
Per the JSON specification, NaN floats cannot be represented natively, so many encoders substitute NaN with null. One would expect PJ to handle null values in JSON input as NaN for compatibility with this common case. Instead, we get this:

[image]

The JSON doc I used is:

```json
{
  "1252": {
    "timestamp": {
      "microsecond": 0
    },
    "value": {
      "current": {
        "ampere": null
      },
      "voltage": {
        "volt": 24.852617263793945
      }
    }
  }
}
```

facontidavide, 2022-07-25:
I updated the JSOn parse and I think this fixed the issue

pavel-kirienko, 2022-08-05:
@facontidavide Thanks! But the issue is still there though :D

facontidavide, 2022-08-05:
sorry to hear that. I will try to address it in the next release

facontidavide, 2022-08-10:
I have just tried and it works just fine
[image]

Closing not being reproducible

commit a61402ec (2022-07-25): update nlohmann json to fix #640
closed 2022-08-10 by facontidavide