PlotJuggler/PlotJuggler #642: vertically-split plots become decoupled when one is dragged

v4hn, 2022-05-09:
encountered with version 3.4.3 (ROS noetic)

Steps to reproduce:

split plots vertically and add related time series:
[image]

zoom in to one plot, the other plot aligns on the x axis:
[image]

drag top plot with middle mouse button. X axis of bottom view should follow, but does not:
[image]

facontidavide, 2022-05-14:
> drag top plot with middle mouse button

I don't think you can drag and drop with middle mouse. I surely can't on my computer

v4hn, 2022-05-14:
I'm sure I can (and I actually don't remember a time I couldn't).
Here's a screen recording: [MP4/H264](https://tams.informatik.uni-hamburg.de/people/goerner/files/plotjuggler_drag_axis_desync.mp4) / [WEBM](https://tams.informatik.uni-hamburg.de/people/goerner/files/plotjuggler_drag_axis_desync.webm).
Notice how
- the cursor changes to a "drag" shape when middle mouse button is pressed
- dragging leaves the other plot not moving
- zooming in/out realigns the two plots
- by dragging I can actually move past the end of the data series (towards the end of the video)

It seems you/plotjuggler were just not aware of the callbacks for this?
I just made this screencast with Qt5.15.2 / qwt6.2.0 / ROS noetic / PlotJuggler commit d1544ffa79d8556058597a22365b761126c3313f (almost current main), but I can definitely reproduce (at work) with a standard Ubuntu 20.04 / ROS noetic released plotjuggler.

Bartimaeus-, 2022-05-28:
Am able to reproduce the bug on Windows with commit https://github.com/facontidavide/PlotJuggler/commit/e4ee67b9421adcb7a0e87e9b22d7d194afa83932:

https://user-images.githubusercontent.com/2954254/170838692-d98cdc11-54a9-4bc2-9d9d-67b395d9b62f.mp4

facontidavide, 2022-08-05:
fixed

v4hn, 2022-08-08:
Thank you so much! :bouquet: 

I am wondering whether it might feel better to move all plots around already on drag and not just on drop for "smoother" user experience, but I absolutely love the fix as is, so I will close this issue. :sauropod:

closed 2022-08-08 by v4hn