PlotJuggler/PlotJuggler #643: [usability] plot all positions of joint set

v4hn, 2022-05-09:
Hey @facontidavide , I promised you in person to report this at least a year ago. Guess I finally manage...
Observed on ROS noetic / PlotJuggler 3.4.3 and earlier.

Before the "big" revamp of the GUI it was straight-forward to select all positions for a set of joints, by searching for a substring of the joint and "pos", selecting all entries as a group and dragging them to the plot.

With the newer GUI this is quite tedious. I can still search for, e.g., "r_ pos", but this is the result:
[image]

Now I have to double click all joints I'm interested in to finally select pos in each of them:
[image]
Group selection via holding shift works, but you actually have to click "position" somewhere because the joint names are not actually selectable *and* the joint entries have to be unfolded already. With many joints and trying to plot position/velocity/effort for them this is very tedious.

facontidavide, 2022-05-14:
try unchecking this

[image]

closed 2022-05-14 by facontidavide

v4hn, 2022-05-14:
:+1: Thank you. I didn't consider you actually made it optional.
Minor issue: Disabling this option with a loaded dataset will not change the layout of the new dataset and adding the same dataset again will leave the screen with a mixture of tree-structured items and a plain list in the view on the left.